ssilverman/QNEthernet #2: [Enhancement] WriteFully for a dynamic String.

Electrolyt, 2022-01-30:
Hi Silverman,

you made a really impressive ethernet library for the teensy 4.1.
It's fantastic. The library is much more stable like old once for the Teensy 3.x.
Is it possible to add a new function for a dynamic string 
to transfer them completely with the WriteFully-command?
I will use it for my web-server.
Example:

WebClient.writeFully("<s40>");
WebClient.print(String (IO.sDC_Current[1]));
WebClient.writeFully("</s40>\r\n");

Best greetings

ssilverman, 2022-01-30:
Thank you.

I’m not completely certain what you’re asking for. There’s already functions for transferring both bytes and chars. What is `IO.sDC_Current[1]`?

ssilverman, 2022-01-30:
My intent for the `writeFully()` functions was to match the `write()` functions. It is my preference not to provide too many convenience functions because it’s often straightforward to convert to arguments needed by one of the existing forms, for example, `String::c_str()`.

What happens if you try this:
```c++
WebClient.print(String(IO.sDC_Current[1]).c_str());
```

I may change my mind in the future, but I first want to see the prevalence of needs for `writeFully(String)`.

Electrolyt, 2022-01-30:
oh, i'am sorry for my incomplete description. 
It is only a variable like integer that will converted into a string.
my meaning: 

WebClient.writeFully("");
WebClient.print(String (intVar));
WebClient.writeFully("\r\n");

About of the String::c_str(). I dind`t know about that. This did the trick. 
I will use:

WebClient.writeFully((String (intVar)).c_str());
I will rewrite my sketch to test them.

Thanks...

ssilverman, 2022-01-30:
I’m curious, why are you writing an empty string in the first line?

Electrolyt, 2022-01-30:
WebClient.writeFully(""); -> this isn`t realy in my sketch. i modified this only for the posting.
i know the command isn`t necessary.

ssilverman, 2022-01-30:
I’m sorry, I mistyped my comment above. I meant this:

```c++
WebClient.writeFully(String(IO.sDC_Current[1]).c_str());
```

Electrolyt, 2022-01-30:
Yes, we mean both the same.
Thank you. Can i ask you in the future again two things.
I haven`t now time for that.
I will leave this issue open, just for communication okey ?

ssilverman, 2022-01-30:
If it’s a different subject from “writeFully(String)”, I’d prefer to use a different issue. Maybe the questions can help others, but I think categorization is important.

closed 2022-02-06 by Electrolyt