ssilverman/QNEthernet #6: System hang when connecting to client that doesn't exist.

videojedi, 2022-02-15:
### Discussed in https://github.com/ssilverman/QNEthernet/discussions/4

<div type='discussions-op-text'>

<sup>Originally posted by **videojedi** February 14, 2022</sup>
Hi, 
Firstly, many thanks for your hardwork, it's really helping me with a project.

However, I'm a little stuck with a client connection to a server that isn't there. My teensy just hangs with the following message

`Assertion "tcp_input: pcb->next != pcb (before cache)" failed at line 269 in C:\Users\admin\Documents\Arduino\libraries\QNEthernet\src\lwip\tcp_in.c
`

I'm just just initiating the connection with

`if (client.connect(setupIP.rtrIPaddr, setupIP.rtrPort)) {
blah
}
else {
Serial.println("connection failed!");
}`

Even the watchdog, is unable to recover the system hang.

Any thoughts greatly appreciated..</div>

ssilverman, 2022-02-15:
Thanks for filing this. I don't see the error.

Running without commenting `PollHardware()`:
```
Values Loaded

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:ab:cd:ef
DHCP
=========================
IP Address = 192.168.1.120
.get Connection failed!
Resetting Soon..
.get Connection failed!
.get Connection failed!
.get 
```

Running with `PollHardware()` commented out and connecting via the server:
```
Values Loaded

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:ab:cd:ef
DHCP
=========================
IP Address = 192.168.1.120
..Resetting Soon..
.............Accepted new websockets client at index 0
Sending {"protocol":"SWP-08","rtrIPaddr":"255.255.255.255","rtrPort":65535,"ipAddr":"192.168.1.120","subnet":"255.255.255.0","gw":"192.168.1.1","dhcp":true}
..........
```

ssilverman, 2022-02-15:
Which versions of Teensyduino and _QNEthernet_ are you using?

videojedi, 2022-02-15:
> Which versions of Teensyduino and _QNEthernet_ are you using?

Teensyduino 1.56 and QNethernet 0.13.0

`Using library QNEthernet at version 0.13.0 in folder: C:\Users\admin\Documents\Arduino\libraries\QNEthernet 
Using library EthernetWebServer at version 2.0.0 in folder: C:\Users\admin\Documents\Arduino\libraries\EthernetWebServer 
Using library Functional-Vlpp at version 1.0.2 in folder: C:\Users\admin\Documents\Arduino\libraries\Functional-Vlpp 
Using library WDT_T4-master at version 0.1 in folder: C:\Users\admin\Documents\Arduino\libraries\WDT_T4-master 
Using library WebSockets2_Generic at version 1.10.0 in folder: C:\Users\admin\Documents\Arduino\libraries\WebSockets2_Generic 
Using library ArduinoJson at version 6.19.1 in folder: C:\Users\admin\Documents\Arduino\libraries\ArduinoJson 
Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\EEPROM 
Using library Entropy in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Entropy (legacy)
`

ssilverman, 2022-02-15:
I just tried with _QNEthernet_ v0.13.0 and get the same result. I don't see this error.

videojedi, 2022-02-15:
So, just checking, with the PollHardware() checking a host that doesn't exist, you can happily refresh the webpage without issue?

I will try a complete fresh install of libs and Teensyduino and build again.

Many thanks for taking a look, really appreciate it.

ssilverman, 2022-02-15:
I also just tried changing the IP address and port to something I know doesn't exist: 192.168.1.100:7000. I uncommented `PollHardware()` and again see the following, with a continuous reboot:
```
Values Loaded

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:ab:cd:ef
DHCP
=========================
IP Address = 192.168.1.120
.get Connection failed!
Resetting Soon..
.get Connection failed!
.get Connection failed!
.get 
```
...and then it reboots and shows the same thing.

I don't see the lwIP error.

ssilverman, 2022-02-15:
No, I can't refresh the webpage because it continuously reboots after the above messages.

ssilverman, 2022-02-15:
Is it possible that you see the lwIP error when the connection to the configured IP address and port is successful? Do you have some code I could set up that would allow the Teensy to connect and communicate with the "hardware"?

ssilverman, 2022-02-15:
Update: Something changed (I didn't change the code, it merely rebooted and now it's staying "up"), and I now see this:
```
Values Loaded

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:31:1a
DHCP
=========================
IP Address = 192.168.6.122
.get Connection failed!
.get Connection failed!
.get Resetting Soon..
Connection failed!
.get Connection failed!
.get Connection failed!
```
...with `.get Connection failed!` continuously being printed and no continuous rebooting. Still no lwIP message.

videojedi, 2022-02-15:
The 'hardware' is a video router running SWP-08 protocol. But when I'm connected to it with correct IP and Port, I have never seen the lwIP error.

Just trying to record a video of my screen....

ssilverman, 2022-02-15:
So it sounds like when the connection can't be made then you see the error? I'd like to make my setup as close to yours as possible. What is your configured IP address and port?

ssilverman, 2022-02-15:
Update: I still can't connect to the HTTP server when it's not rebooting and printing `.get Connection failed!` continuously.

videojedi, 2022-02-15:
> So it sounds like when the connection can't be made then you see the error? I'd like to make my setup as close to yours as possible. What is your configured IP address and port?

192.168.9.124 port 8000 for the 'hardware', teensy is 192.168.9.117

videojedi, 2022-02-15:
https://youtu.be/7c-ZITIZqHA

ssilverman, 2022-02-15:
Thanks for the video. I can't reproduce what's on that because the Teensy continuously reboots after the following message and I can't make an HTTP connection:
```
Values Loaded

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:ab:cd:ef
DHCP
=========================
IP Address = 192.168.1.120
.get Connection failed!
Resetting Soon..
.get Connection failed!
.get Connection failed!
.get 
```

ssilverman, 2022-02-15:
I changed the watchdog call at the end of `loop()` to:
```c++
//  feedDog();
  wdt.feed();
```

And now I can connect, but it's slow to respond.

I reload the page and see the following, still no lwIP error:
```
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
Connection closed
.get Connection failed!
.get Connection failed!
.get Connection failed!
Accepted new websockets client at index 0
Sending {"protocol":"SWP-08","rtrIPaddr":"192.168.1.100","rtrPort":8000,"ipAddr":"192.168.1.120","subnet":"255.255.255.0","gw":"192.168.1.1","dhcp":true}
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
```

videojedi, 2022-02-15:
I've removed the watchdog, in case that was causing the reboots

ssilverman, 2022-02-15:
Given that I can't reproduce this, the only thing that appears different to me between our setups is the fact that you transition from a working IP address to one that isn't working. Do you still see the lwIP error if you set the address to something non-connectable and then reboot the Teensy?

My current guess is that there's some sort of memory corruption happening somewhere and an invalid object is being used or reused. @khoih-prog do you have any thoughts on this?

ssilverman, 2022-02-15:
@videojedi Let's test my theory.

In your WebSockets2_Generic library source (in the Arduino library folder, wherever that is for you), in `src/Tiny_Websockets_Generic/network/Teensy41_QNEthernet/Teensy41_QNEthernet_tcp.hpp`, replace this:
```c++
        TcpClient* accept() override 
        {
          if (server == nullptr) 
          {
            return nullptr;  // Watch out for this!
          }
          
          auto client = server->accept();
          
          return new EthernetTcpClient(client);
        }
```

With this:
```c++
        TcpClient* accept() override 
        {
          if (server == nullptr) 
          {
            return new EthernetTcpClient(EthernetClient());
          }
          
          auto client = server->accept();
          
          return new EthernetTcpClient(client);
        }
```

Let's see if that helps. Crossing fingers.

khoih-prog, 2022-02-15:
Hi @videojedi 

Thanks for the tag and the investigation so far, @ssilverman

I couldn't duplicate the issue and wonder if there is anything wrong in your installation, settings and/or board.

It's really hard and almost impossible to find out where the problem is, w/o being able to duplicate.

The terminal just goes forever without any issue or reset

```
.get Connection failed!
...
.get Connection failed!
```

[image]

BTW, did you try to use the [**Teensy Patches**](https://github.com/khoih-prog/EthernetWebServer#2-for-teensy-boards). I actually don't remember why and which issue the Patches fix now.

khoih-prog, 2022-02-15:
HI @videojedi 

It's also better to fix this `warning: control reaches end of non-void function `

```
IPAddress StringAddress2IP(const char * string) {
  IPAddress sendVal;
  if (sendVal.fromString(string)) {
    return sendVal;
  }
}
```

https://github.com/videojedi/js3debug/blob/8148fe7fe13510630ac9f39d3c2fd911522b303d/JS3debug.ino#L167-L172

ssilverman, 2022-02-16:
@khoih-prog any thoughts on `accept()` possibly returning NULL and what callers may do with that? (See: https://github.com/ssilverman/QNEthernet/issues/6#issuecomment-1040854877)

khoih-prog, 2022-02-16:
I think the caller has to be aware of and to check for  `NULL`  return because of the

```
return new EthernetTcpClient(client);
```

ssilverman, 2022-02-16:
What I’m really asking is: are you certain that all callers (in all your libraries that use it) of `accept()` check for NULL properly in all cases. I’m not referring to any callers; I’m specifically referring to your own libraries that @videojedi is using, not any arbitrary caller.

khoih-prog, 2022-02-16:
> What I’m really asking is: are you certain that all callers (in all your libraries that use it) of accept() check for NULL properly in all cases. I’m not referring to any callers; I’m specifically referring to your own libraries that @videojedi is using, not any arbitrary caller.

If so simple, it can be duplicated and fixed right away.

ssilverman, 2022-02-16:
It was a sincere question. :)

I haven’t looked through the source too much, which is why I asked. I’m trying to rule out “simple” but unintentional bugs.

videojedi, 2022-02-16:
Really appreciate the help.

Just catching up on comments and suggestions.

I can confirm I have loaded the teensy patches for stream.cpp and stream.h

I have fixed the compiler warning error.

And I've tried the return new ethernetTCPClient in teensy41_qnethernet_tcp.hpp, 

Hasn't made a difference, so I think its time to try a compile on a new clean install.

I'll come back to you.... thanks again

videojedi, 2022-02-16:
I've just uploaded a compiled binary from my IDE and libs, if you get a chance, maybe you could upload this to a teensy4.1. This would at least confirm you can see the same issue, and point towards my install being the issue.... if not, I guess it could be hardware...

ssilverman, 2022-02-16:
@videojedi To where did you upload this binary?

videojedi, 2022-02-16:
Should be in the GitHub repo of the demo app

videojedi, 2022-02-16:
It should continue with failed connection log entries. 

My log runs at the expected once a second speed. Until I refresh the web page, then get an immediate assert message and everything stops.

ssilverman, 2022-02-16:
My mistake — the lines do keep repeating; I had a second serial terminal open. But the page still takes well over a minute to load, and not all the content is loaded, eg. the CSS. I also still don't see the lwIP error.

videojedi, 2022-02-16:
Just about to order a new teensy. 
Don't know enough about the nuts and bolts of what's going on handling the connections. But assume some timeouts are getting in the way of the web server handling requests? I had added connectionTimeouts of 100mS but didn't seem to help.

videojedi, 2022-02-16:
Very occasionally, I think I have seen slow connections like you've seen, maybe get the HTML but not the CSS. Mostly just get the lwIP error, as below.

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:df:97
DHCP
=========================
IP Address = 192.168.9.113
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
Assertion "tcp_input: pcb->next != pcb (before cache)" failed at line 269 in C:\Users\admin\Documents\Arduino\libraries\QNEthernet\src\lwip\tcp_in.c

ssilverman, 2022-02-18:
@videojedi when it hangs, are you printing `CrashReport`?

Try adding this at program startup, just after initializing Serial:
```c++
if (CrashReport) {
  Serial.println(CrashReport)
}
```

if it’s a true hang, that will print something and allow you to find where the crash occurred.

videojedi, 2022-02-18:
Had a crash report immediately at startup..

`Values Loaded
CrashReport:
  A problem occurred at (system time) 18:7:3
  Code was executing from address 0x0
  CFSR: 1
	(IACCVIOL) Instruction Access Violation
  Temperature inside the chip was 48.14 °C
  Startup CPU clock speed is 600MHz

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:df:97
DHCP
=========================
IP Address = 192.168.9.114
.get Connection failed!
.get Connection failed!
`

But not when refreshing the page. Just the assert. Interestingly, the ping command I had running, carries on responding, although a bit intermittent. See video

https://youtu.be/dfoqzc-HK7c

ssilverman, 2022-02-18:
It prints the report from a crash that occurred before startup. When a crash happens, it saves the info for the next time you print the report. This means that you did not experience a crash at startup; you’re only seeing the saved result.

My question is: when you see the lwIP assert and the board restarts (I’m assuming that the assert is followed by a crash and restart), do you see a crash report then?

videojedi, 2022-02-18:
I don’t think the board fully restarts. As it’s still sort of responding to ping. The app certainly doesn’t start again until I manually restart it.

videojedi, 2022-02-18:
Here is log from recent

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:df:97
DHCP
=========================
IP Address = 192.168.9.139
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
Assertion "tcp_input: pcb->next != pcb (before cache)" failed at line 269 in C:\Users\admin\Documents\Arduino\libraries\QNEthernet\src\lwip\tcp_in.c

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:df:97
DHCP
=========================
IP Address = 192.168.9.139
.get Connection failed!

Starting AdvancedWebServer on TEENSY 4.1 using QNEthernet
EthernetWebServer v2.0.0
MAC = 04:e9:e5:10:df:97
DHCP
=========================
IP Address = 192.168.9.139
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!
.get Connection failed!

Each time it's a manual reboot.

Should have 2 brand new Teensy4.1s tomorrow.

ssilverman, 2022-02-18:
This is a frustrating bug, especially since I can’t reproduce it. Just keep hunting, I guess. I’ll endeavour to find what code paths can lead to that assert.

khoih-prog, 2022-02-23:
I guess that the issue has been resolved by the **OP** replacing the defective board ???

videojedi, 2022-02-23:
Sorry for delay. New boards only arrived yesterday. I tried both last night and the same issue remains. Also tried a new build from a new machine (a Mac this time) again the same issue.

ssilverman, 2022-02-23:
I think the best approach now would be starting with the minimal possible app, verify that it doesn’t fail, and then keep adding very small pieces, one at a time, until you see the error.

khoih-prog, 2022-02-23:
I totally agree with @ssilverman  approach. It's something very elusive in your code @videojedi. Without reliable duplication, I'm afraid we can't do anything to isolate / fix.

videojedi, 2022-02-23:
Will do. I’m wondering if the EEPROM code is getting in the way.

ssilverman, 2022-03-04:
Just following up. I haven't been able to reproduce it, and was going to suggest that you open a new issue if you can come up with a minimal code example that demonstrates the problem. I'd like to close this at this point.

videojedi, 2022-03-04:
Hi, 

Work kinda got in the way this week, although I've made some progress, but need to verify what I've changed and the circumstances that led to an improvement.

Briefly, the target device is now different and I'm on a different network to the one I've been using previosuly. Also I now only poll the target device once every 30secs, once one attempt has failed. Returning to once every second when the target device is available again.

The result is, I haven't seen another assertion error. And the webpage has time to load during the 30secs before the next poll.

Hopefully I will get some more testing time this weel, but please feel free to close this. 

Many thanks for your help.

Richard

ssilverman, 2022-03-22:
Thanks for the update. Closing.