ssilverman/QNEthernet #7: Compiling with M1 Mac

Rothemich, 2022-03-12:
Has anyone successfully compiled with this library in VSC with PlatformIO on an M1 Mac?  I'm getting a warning:  

> Warning! Could not find the package with 'ssilverman/QNEthernet @ ^0.14.0-snapshot' requirements for your system 'darwin_arm64'

VSC=1.65.2
PlatformIO=2.4.2
QMEthernet=^0.14.0-snapshot
board=teensy41

ssilverman, 2022-03-12:
What does your platformio.ini file look like?

For lib_deps, I would use `https://github.com/ssilverman/QNEthernet.git` instead of `ssilverman/QNEthernet` (and no version).

Note that because the version stays the same in the downloaded library (in this case, 0.14.0-snapshot), the system doesn’t update it until the version number changes. It won’t change on GitHub until I update to v0.14.0. To get the very latest, delete `.pio/libdeps/teensy41/QNEthernet` and rebuild.

ssilverman, 2022-03-22:
Is this still an issue for you?

Rothemich, 2022-03-23:
I don't **_THINK_** so.  The inclusion of the git reference in lib_deps did appear to get me past the package issue.  Now I believe I have a set of other library mismatches that are causing issues, but are presumably entirely unrelated.  In honestly, I've only had couple hours since posting to begin sorting it out, but I'll FUP once I get back to it.  Thank you for your response.

closed 2022-03-23 by Rothemich