ssilverman/QNEthernet #9: Entropy.h: No such file or directory

piranha771, 2022-04-14:
I installed the lib as a dependency from [https://github.com/khoih-prog/EthernetWebServer](url) inside of PlatformIO
But when building I get the following error:

`.pio\libdeps\teensy41\QNEthernet\src\QNEthernet.cpp:12:21: fatal error: Entropy.h: No such file or directory`

I can't find any reference in the project to a lib called "Entropy", so I can't install the dependency manually.
Please have a look into this.

ssilverman, 2022-04-14:
Are you including QNEthernet.h?

ssilverman, 2022-04-19:
The solution is one of:
1. Include `<QNEthernet.h>` in your project, or
2. In `platformio.ini`, set `library_ldf_mode = deep` (or `deep+`).

By default, PlatformIO only looks one level deep for dependencies, and the Entropy library is a dependency of _QNEthernet_. Since _QNEthernet_ is itself a dependency of one of the project’s library dependencies, the system won’t find the Entropy library without one of the above two steps.

closed 2022-04-19 by ssilverman

piranha771, 2022-04-20:
Hi, sorry for the late reply.

I can confirm that `library_ldf_mode = deep` does indeed fix the problem.

Thank you for your quick response.