ssilverman/QNEthernet #11: Need to disable printf...

wwatson4506, 2022-04-22:
Hey Shawn,
I just started working with QNEthernet for the purpose of updating my FTP client and server libraries on the Teensy 4.1. Ran into an issue with using printf() in those libraries. I have am IO library that uses a POSIX style OS and for some reason printf() does not work when used with QNEthernet. It works with NativeEthernet. I guess there is a conflict of some sort. So to the point, I want to use your Ethernet library and was wondering if there was a way to disable printf() in QNEthernet?

ssilverman, 2022-04-23:
Try defining the `QNETHERNET_WEAK_WRITE` macro in your build. Eg. add a `-DQNETHERNET_WEAK_WRITE` build flag. That will make _QNEthernet_’s version of `_write()` weak.

wwatson4506, 2022-04-23:
Man that was quit,
Thanks for responding. I will check it out. Seems a lot of things are going on with FTP, WIFI and the right now:)

closed 2022-04-23 by wwatson4506