ssilverman/QNEthernet #13: Possibility of adding a new MAC-PHY

vChavezB, 2022-05-03:
I am currently working on a research project where we want to develop an Arduino framework for the latest ethernet PHY, [10BASE-T1L](https://www.analog.com/media/en/technical-documentation/tech-articles/enabling-seamless-ethernet-to-field-with-10base-t1l-connectivity.pdf). As a brief context, 10BASE-T1L allows a single pair wire connection up to 1KM distance and PoDL (class 15 60W).

We want to use LwIP and integrate it into the Arduino platform. We have already [ported](https://github.com/vChavezB/lwip-Arduino) LwIP to Arduino as a standalone library that exposes the API. However, we are missing the high-abstraction layer that could be used easily with the Arduino platform.

As this is something you already did in this project I would like to ask how difficult would it be to change the MAC-PHY low-level implementation from the Teensy. Is the low-level Ethernet driver tightly coupled? or could we easily replace the functions to have a generic interface that connects to lwIP. and the rest of your application layer. 

We could try to use this project as a reference...but if we could tinker easily with the low-level driver implementation then we wouldn't need to reinvent the wheel.

After a brief review, I see for example that the low-level driver implementation is done in 

- [`enet_init`](https://github.com/ssilverman/QNEthernet/blob/fa3369284a90f8e26c0a9c82f395afecdf6f1642/src/lwip_t41.h#L40)
- [`enet_deinit`](https://github.com/ssilverman/QNEthernet/blob/fa3369284a90f8e26c0a9c82f395afecdf6f1642/src/lwip_t41.h#L47)
- [`enet_proc_input`](https://github.com/ssilverman/QNEthernet/blob/fa3369284a90f8e26c0a9c82f395afecdf6f1642/src/lwip_t41.h#L55)
- [`enet_poll`](https://github.com/ssilverman/QNEthernet/blob/fa3369284a90f8e26c0a9c82f395afecdf6f1642/src/lwip_t41.h#L58)

Best regards,
Victor

ssilverman, 2022-05-03:
It shouldn't be that difficult. See, for example, Paul Stoffregen's `lwip_k6x` files here: https://github.com/PaulStoffregen/teensy41_ethernet/tree/master/t41ether/lwip/src

There might be some fine tuning you have to do.

See also what `EthernetClass` does and how it uses the low-level API.

vChavezB, 2022-05-07:
Ok nvm, I found out that the API for the MDNS app changed. The parameter `dns_ttl` in function `mdns_resp_add_netif` is not needed anymore as the value is provided internally by the lwip stack.