ssilverman/QNEthernet #26: Teensy 4.1 freezing, TCP connecting during

megax, 2023-01-12:
Hi,

QNEthernet version: 0.17.0
I have the following problem. Given a router and the library you made.

(Important, I only experienced this on this router, not the other one.)
Device: Teltonika RUT955 https://teltonika-networks.com/product/rut955/
(Firmware is the last.)

The error is occuring when I restart the router and it starts, then the mobile network is restored, after that I am trying TCP connect while the Teensy got frozen. (See attached log)
I restart Teensy, if the internet is stable, there are no problems. The connection is successful. Not even after removing the Ethernet cable (and then reconnecting).
More information, a VPN network is running on the router, in the direction of an external server, through which Teensy is also connected.
I also used LWIP_DEBUG . (Now afterwards, in the case of the attached test code and in the case of the log, it was not yet present). 

Based on this, I experience the error discussed here, in this case the error message: https://github.com/ssilverman/QNEthernet/discussions/4
`Assertion "tcp_input: pcb->next != pcb (before cache)" failed`

Unfortunately, I can't figure out why. I tried to modify the code (Also the lib you provided to replace close()->abort() with the description here: https://github.com/ssilverman/QNEthernet/issues/15) 

Unfortunately, it didn't change anything.
I also experience this, right before the freezing in the message package:

```
tcp_pcb_purge
tcp_pcb_purge: data left on ->unacked
```

It is also not clear why the error occurs. 

The VPN is not an absolute source of error because there is also a RUT950 https://teltonika-networks.com/product/rut950/ router with the same settings, yet I do not experience this. 

I tried to reproduce it in vain. Even if it's the router's fault, I'm not sure if this is the correct behavior for the code to do in response.
Teensy doesn't freeze in a single case (or not always) if I extend the check times for retries to 50-60s, but that's not a solution for me. 

I would use it for a GPS-based system, for which it is important to restore the external connection as quickly as possible for RTK accuracy. (On this my system worked with another microcontroller so far, particle photon 1, but Teensy is a better platform for future development/evolution, so I would move my code here. I like your lib best as a TCP implementation and it works well only with the TCP connection. NativEthernet lags many times when the connection breaks, it is simply unusable.)
I want to use the type of router written above.

Can I ask for help to solve this problem? Possibly, if you provide switches for debug, I'll convert it and attach it if that helps.

[Teensy freezing.txt](https://github.com/ssilverman/QNEthernet/files/10405337/Teensy.freezing.txt)
[image]
[image]
main.cpp (test code) [main.txt](https://github.com/ssilverman/QNEthernet/files/10405342/main.txt)

ssilverman, 2023-01-12:
Thanks for the report. I’ll have a look.

ssilverman, 2023-01-12:
Just off the top of my head, do you think this is related:
https://github.com/ssilverman/QNEthernet/blob/master/README.md#on-connections-that-hang-around-after-cable-disconnect

I just did a quick trace through the code and I wonder if this is some lwIP bug.

For reproduction, is this happening every time the router starts or you can’t reproduce it at all and you’ve only seen this once? Your text seems to imply both and I’m not certain if you can reproduce it or not.

If you can reproduce it, I could make a version of the library with the latest lwIP master (the library is currently on 2.1.3) that you can try. Let me know.

ssilverman, 2023-01-13:
I just pushed a `lwip-2.2.0-snapshot` branch. Note that it includes a bunch of other changes on top of v0.17.0. Can you please try this and see if it fixes the issue? In the meantime, I'll have a look at your code you added to the report (_main.txt_).

ssilverman, 2023-01-14:
A few minor notes to the code as I'm diving in:
1. Don't need to specify a MAC address in _QNEthernet_. Instead, change your `begin()` line to:
   `Ethernet.begin(ip, maskk, gatway, gatway)`.

   You can retrieve and print the MAC address with:
   ```c++
   uint8_t mac[6];
   Ethernet.macAddress(mac);
   print-the-mac-address();
   ```

   Each Teensy 4.1 has a built-in MAC address.
2. I know that it's commented out, but you shouldn't call `Ethernet.begin(...)` from inside an Ethernet listener. Once Ethernet is started, it shouldn't usually need to be started again.
3. In _QNEthernet_, there's no need to check for hardware status (it will always return `EthernetOtherHardware`). These Arduino-style API's were designed to be run on a range of possible boards, so these kinds of functions are useful for these unknowns.
4. If you check `Ethernet.linkStatus()` before the polling has had a chance to spin up, it will always return `LinkOFF`, so checking it at program startup won't give you useful information. Internally, `Ethernet.loop()` is called frequently and the link status is only polled about 8 times a second or so. Using the listener like you've done is, in my opinion, a good way to monitor the link status. `Ethernet.begin(...)` is what causes the link listener to be called.
5. If you always use the same port for `EthernetServer`, then just declare it (where you have it at the top) as:
   `EthernetServer myserver(3456);`
   If you want to change the port, I've extended the Arduino-style API to have `begin()` functions that take a listening port as an argument. i.e. declare the server as `EthernetServer myserver` and then call `myserver.begin(some-port-value);`.
6. Instead of calling `Ethernet.linkStatus()` in the main loop, set some boolean state from inside the `onLinkState` listener and check that instead.
7. You may already know all these notes because you may need to run similar code on other systems, but I'm including them here for posterity and any future readers. :)
8. I don't see anywhere you're calling `myserver.accept()` to actually accept connections? Have you seen the _QNEthernet_ server examples?

ssilverman, 2023-01-14:
Question: How are you enabling lwIP's internal `printf` for the debug output? Are you setting
`stdPrint = &Serial` somewhere after `Serial.begin(115200)`?

megax, 2023-01-14:
Thank you very much for the additional information. I would respond to them in sequence. :)

1. First, I used NativEthernet's code, its API structure appears in my sample code. To my real code, I already did this according to the description you provided. I'm also glad that you have your own mac address, I wouldn't want to deal with/generate that either.

2. I just tried there, that's why it's commented. While analyzing your code, I realized that it makes no sense to use it like this.

3. I also took this from NativEthernet, I have already deleted it.

4. In the loop part of my code, it was `WiFi.ready()` until now. Replaced everywhere with Ethernet.linkStatus(). In addition to this invitation, it is only there if there is an error, to break the connection. I tried to solve the freeze I was experiencing with it. It was a bit of a test, but it's not bad in the event of a network outage. The first LinkOff is understandable, it's not an absolute problem for me if it doesn't cause other regressions.

5. Although I use it, it is a global variable in my main code. I only used it here as an API test.

6. I have a write function in many places, actually a tablet is connected to my code. On the outside, I reach an ntrip server with TCP protocol, and another TCP protocol transmits external data and receives data. Other microcontrollers tend to freeze if I haven't checked them in time, for certain code sections. Is it okay if I continue to check this with `WiFi.ready()`?

7. Ok, thanks for the info :)

8. Here in this example code, it was only a connection test. In other words, can I reach it, for example, with a tablet or with the Windows telnet command. My regular code is written normally, and handles it stably. My goal here is primarily API compatibility, but I also used the modifications you wrote. I just need time to make it "nice". I'm still analyzing it, but at the moment it works fine. For example, Nagal's algorithm is inactive. It’s disturbing, but as described, I set it to false for every new connection.

`stdPrint = &Serial`

Yes, I'll define it later. Has it significance? Would it delete Serial if I set it first?

I can only test the code on Monday. The router is located in a machine that is working and in use, and is also used on weekends. (prototype, but used) I will write a result afterwards. I test only with lwip update and then with the modified library. I am attaching the log about them.

Will the first Debug option be enough for testing? or should I change anything?

Thank you in advance for your reply.

I'm attaching the test code which has been motified by the description. 
[main.txt](https://github.com/ssilverman/QNEthernet/files/10417473/main.txt)

ssilverman, 2023-01-16:
I was just wondering what you did to enable the debug output when you turned on debug in lwIP. (I’m not following your answer.) It uses `printf` internally.

I’m still looking at the code. I tried a few incarnations of my library and, with some later changes, I can’t get your code to fail. So I’m in the process of figuring out which change seemed to fix it. Before that, the code was crashing the Teensy. I’m curious about your results when you test with the two newer branches.

megax, 2023-01-16:
I tested it, I still experience the following error. Both with test code. I am attaching the log below. (from `lwip-2.2.0-snapshot` branch, same result for another)

[log.txt](https://github.com/ssilverman/QNEthernet/files/10424070/log.txt)

ssilverman, 2023-01-17:
Some comments about some things I noticed:
1. Change this:
   `EthernetServer myserver = EthernetServer(3456);`
   to this:
   `EthernetServer myserver(3456);`
  I've removed the copy constructor for the upcoming library v0.18.0 and that first way won't compile.
2. Point #1 leads me to believe you didn't correctly try the `lwip-2.2.0-snapshot` branch because your latest _main.txt_ code shouldn't compile with that branch.

Question: I know it may not seem relevant, but I'm trying to figure something out about your original state. When you enabled debug, how exactly did you get output? lwIP debugging uses `printf` under the covers, which means that function has to be wired up because it's not by default. Did you modify the lwIP files to change `printf` to `Serial.printf` or something? That's why I was asking about `stdPrint = &Serial`. If you didn't do that, and if you didn't define your own `_write()` function, then I'm not sure how you got any debug or assertion output.

ssilverman, 2023-01-18:
So I've been testing the latest _main.txt_ where I only changed the IP address information, and I can't get it to fail. Is there anything you can think of that you changed either in QNEthernet or in the Teensyduino core?

I've tried with newer and older QNEthernet versions and also the 1.58-beta3 and 1.57 Teensyduino compilers and cores.

ssilverman, 2023-01-18:
Can you try adding this to `setup()`, after initializing `Serial`:

```c++
if (CrashReport) {
  Serial.println(CrashReport);
  CrashReport.clear();
} else {
  Serial.println("No CrashReport");
}
```

When it crashes and then restarts, it should print something there.

megax, 2023-01-18:
Thanks for your response.

Points 1 and 2 written above are true, but not as you interpret them. I use PlaformIO and I had two separate test codes for the code located in the two branches. This code that I attached is the test code of lwip-2.2.0-snapshot-v0.17.0 due to my mistake. Sorry about that. I am aware that you modified it, I did not call it exactly as you wrote, I just defined it and set the port in setup() with begin(3456). I am also attaching my code for this.

In connection with debugging, I only use what is visible in the code: `stdPrint = &Serial`, nothing else. In connection with this, I only and exclusively enabled the debug define options written above, I use the code you provided for everything else. (I looked at it with debug turned off for the first time, but then because of stdPrint, this error definitely appears: `Assertion "tcp_input: pcb->next != pcb (before cache)" failed`
So, for the sake of testing, I rely on what you give, that there shouldn't be any problems/differences. Teensduino's code is also unchanged.

I will add the `crashReport` section, the only problem is that no crash actually occurs. It stops there one by one. Although debug is not activated, as I can see it printed on `stdPrint`, I suspect that the assert code triggers the `abort()` command. But, teensy doesn't stop and doesn't start again, it just waits there. And nothing happens.

I also looked at the code, after a separate test.
https://github.com/ssilverman/QNEthernet/blob/master/src/lwip/tcp.c#L2186 Here, I think the pcb list is missing the pcb that it would delete afterwards, or at least some memory reference is already NULL. And I think that's why I got an error, which was sent to you. If I change this so that the assert does not run (I turn off the debug), it jumps to what I sent first. But, it's understandable because the addressing is already wrong here, and it causes an error there as well. But, I don't understand what deleted it or what didn't add it. lwip's library is foreign to me, even though I looked it up, I can't find what it could be. That's for sure, this is the status: https://github.com/ssilverman/QNEthernet/blob/master/src/lwip/tcp.c#L397 
It can also be seen in the log. But, it's like you delete it again afterwards and that causes the problem.

Should I provide you access so that you can view the code yourself in a production environment? (with remote desktop) I am at a loss, me too, I was hoping the lwip updates would fix it.

[main.txt](https://github.com/ssilverman/QNEthernet/files/10444871/main.txt)

ssilverman, 2023-01-18:
Excellent. I use PlatformIO too! :)

Some side points:
1. Have you seen this thread: https://forum.pjrc.com/threads/71730-How-to-add-a-Teensyduino-(beta)-release-to-PlatformIO
   In it, I describe a way to use the latest Teensyduino 1.58-beta3 with PlatformIO. That's how I've been testing my stuff lately with the latest beta release.
2. You may already know this (this is for future readers if you do), but here's how to specify a specific library branch inside `lib_deps`:
   https://github.com/ssilverman/QNEthernet.git#lwip-2.2.0-snapshot

One very important point (for future readers) is that when you change anything in `lib_deps`, it doesn't always update the downloaded library cache. You need to delete the (for example) `your-project-dir/.pio/libdeps/teensy41/QNEthernet` folder and then it will refresh with the new library dependency.

Sure, I'd love to experiment with your actual code and environment. Feel free to email me your contact information, and I can set up a Google Hangout or something, where we could maybe explore this together.

ssilverman, 2023-01-18:
I'm wondering about something. Is the _main.txt_ file exactly the same as your testing environment? For example, does the failure happen there too, or only in some "larger program"? If there is some larger program, does it use interrupts or threads to do work with the Ethernet stack?

megax, 2023-01-19:
Thank you very much for the information. :)

Yes, I know the upgrade methods to PlatformIO. I haven't updated Teensyduino yet.
Yes, the test environment and the live code execution location are the same. I'm overwriting Teensy and checking it the same way. Although my main code is a little different, when I found the error, I changed it to the same one, and created this test code based on its simplified model. I have a fork, you can see the change in the `yield()` calls. I have an extra thread running to return async with a callback. I changed Particle's `AsyncTcpClient` (https://github.com/rickkas7/asynctcpclient) code to Teensy and `yield()` is structurally different, so if I use TeensyThread's code, it only works that way. But as I wrote, I turned it off when the error appeared, and it behaves exactly as it appears in the test code I attached. 
However, I found something.

Shortly:
It works now and doesn't freeze. But, this is neither "nice" nor a solution (temporarily it is good, but this error may also occur in other people, and I am also testing it)

Longer:
As I wrote in my first post, I have two types of routers (RUT955 is a slightly more advanced version of RUT950, but almost identical). The code worked fine with the RUT950. However, I just updated the firmware because the RUT955 already has a newer version. I tried with a newer version. Now the RUT950 produces the same error. From this I clearly say that something has changed in the implementation of the Teltonika network interface. 

But, during the day I found something strange.
from the `begin()` base, `ip`, `mask`, `gateway` and `dns`. I gave the router's IP to the `dns`, because there were routers where it caused problems when connecting if I left it blank.
This is where it gets interesting.
If I set it to, for example, 1.1.1.1, the error disappears (in part, I'll explain why in a moment), of course it doesn't matter what we set it to, just have it be an external dns address. I believe that for some reason, when creating the mobilenet, it redirects something else and stops at an empty info, which causes the `pcb` to get confused and that's why it writes `NULL`. Based on these, there is a dns-based error.
I tested it, if I take out a sim card, wait and push it in, it happens as soon as it is created. Just when you create a connection and query a path. 

And here comes the but.
The other part of the truth is that if I don't call it like this: [ntrip2.gnssnet.hu](http://ntrip2.gnssnet.hu/) but with its IP address (`IPAddress ip (84, 206, 45, 44)`) it also crashes and I get the same thing. I noticed it in my main code, I was happy that it worked but on the other hand it didn't work either, while I didn't figure out what the difference was. (it was specified with ip in the config) 
So I had an idea.
```
        IPAddress ip;
        if (DNSClient::getHostByName("google.com", ip, 1000)) {
            entry->result = entry->client->connect(entry->ipAddr, entry->port);
        }
```

I am requesting an address that is only available on the world wide web. And that's fixed too. Well, now I'm starting to understand a little what the problem is. I still can't fix the original code. However, based on the first tests, the error ceased. I tried with my live code, reconnecting several times and doing rough tests. I didn't experience any errors. 

I don't know what I'm writing is understandable, I hope this might help in diagnosing the error. Especially because if it's good for me now, it could happen to someone else at any time. And I wouldn't say that I patched the source of the error in the most beautiful way. I am attaching a test code below, I modified the previous main.txt (ip based). I will email you with contact information, but I don't know yet when a good testing time will be. 

I hope we can find something based on this information :)

[main.txt](https://github.com/ssilverman/QNEthernet/files/10460884/main.txt)

ssilverman, 2023-01-19:
I’m finding it hard to follow what you wrote, but here’s the main points I gathered:
1. The assert and freezing occurs when using an asynchronous approach (eg. adapted asynchronous TCP library or TeensyThreads).
2. The problem is related to DNS lookups.
3. The _main.txt_ code isn’t actually the same code as what you were running that had the assert condition. (Per my question in https://github.com/ssilverman/QNEthernet/issues/26#issuecomment-1396233280)

Unless written in a certain way, _QNEthernet_ doesn’t support running in an asynchronous environment.

megax, 2023-01-20:
1. No, the freezing for which I wrote the error ticket is not the cause. You asked if my master code was different. Yes, it's different. However, what I wrote above was a different error, I just mentioned it. The reason I wrote the error note is what main.txt contains. In order to find the error, I also wrote my original code back to your code, completely, removing the async part that I wrote. So, the original problem for which I wrote the error ticket is the cause of the freeze.
It has nothing to do with what I just mentioned. And it can be reproduced with main.txt.
I run it when I test.

2. yes, I assume that, but I don't know why. (See the new main.txt file at the end of my previous comment.) 

3. No, you are misinterpreting it. main.txt is an extract of my main code. Logically, it works the same way. When I get a repair guide, I'll test it. All the logs I attached were made with this. This main.txt triggers the assertion. 

4. At the end of my previous comment, there is a new main.txt.  This contains changes to the dns error.

5. Changing dns to `1.1.1.1` only resolves "host" calls, for example:  [ntrip2.gnssnet.hu](http://ntrip2.gnssnet.hu/)

6. Changing dns does not solve this: `IPAddress ip (84, 206, 45, 44)`

7. For the error in point 6, the last main.txt contains a correction. `testmode = 2`. Then the ip-based code part runs. In that case, that code is also good. If `testmode = 1`, the malfunctioning ip part runs down. It is the same error as before. 

8. Once again, main.txt is a working code, I tested everything with it, the reason is why I wrote the bug ticket. The log and all tests were made with it. With this, the error can be reproduced and that is why I wrote it so we can reproduce the error again. Even when you wrote a correction. Despite this, I also tried it in my other code and got the same result as here with this simpler code. Now, with the changes I sent in the last comment, my other code now works properly. Of course, main.txt also works properly this way. I tested this first, it didn't throw any error, and so I modified the other one also.

ssilverman, 2023-01-23:
Thank you for clarifying.

I’m still a little unclear on what modifications you made to get this to work for you. Would you mind describing them?

megax, 2023-01-29:
`Testmode=2` and `testmode=0` are working (with prerequisite). `Testmode=0` if I set an external address for the dns server (see `1.1.1.1`). `testmode=2` only works if the dns server is set (see `1.1.1.1`). It does not work even with the `testmode=1` dns server setting. I'm getting the same error that I posted in the error post several times. I don't quite understand why this is, but the source of the error may be related to the dns settings. But, I can't confirm that, it just fixed my test code. It still works that way.

ssilverman, 2023-01-29:
Thanks for the clarification.

I noticed that you have a fork of QNEthernet. Are you using that fork or the main repo?

megax, 2023-01-30:
Yes, I made a fork of it. For my main project, I use my own fork due to yield changes. This change does not affect this error ticket. I use your QNEthernet project for testing. But a fork is also the latest or the latest branch of lwip. With the minimal modification I wrote.

The built-in callback is quite good, but it's not suitable for me, I can't have a 100ms loop waiting in my code. That's why I have to remove the dns max 1000ms timeout in this form. It needs a separate thread. If you can possibly solve this later, I hope I will only use your callback code/implementation.

But, as I wrote, this does not affect the error ticket, it is completely based on your code and I tested with it. I don't use any modifications for the tests, there should be no difference.

ssilverman, 2023-01-30:
I've never been able to reproduce the problem, so I can't help diagnose this, unfortunately. I can tell you these notes, however:

First, _QNEthernet_ doesn't support access in an asynchronous manner, so I suspect your original issue was because of the threads. In https://github.com/The-Megax/QNEthernet/compare/ssilverman:QNEthernet:master...master, there's the addition and use of the `DNSClient::yield_thread()` function. Unless you've guaranteed that _all_ accesses via _QNEthernet_ to lwIP properly synchronize lwIP's core, all guarantees go out the window. You do say that you see the issue with the stock _QNEthernet_ distribution, so I hear that point, but I can't reproduce it, sadly. I've tried with an updated library, v0.17.0 of the library, and also with the latest lwIP master changes without success.

Second, there is a non-blocking version of DNS lookup. See `DNSClient::getHostByName(const char *hostname, std::function<void(const ip_addr_t *)> callback)`. That will return `true` for valid arguments, and then sometime in the future, the function will be called when the DNS lookup completes. You could use standard state machine techniques to, say, set some status flag in the function that you'd check in your main loop. See `QNDNSClient.h` for more information.

In summary, you can solve this without threads. `DNSClient` can be used in a non-blocking way.

megax, 2023-01-30:
1, As I wrote, yield() caused something else, specifically a complete crash (The modification I wrote completely solves the error). Unfortunately, it is not related to this, and unfortunately it is not the cause of my basic problem. Even the plain main.txt that I attach always fails, I use your code entirely when I work with it. Async does not cause any problems, actually, another connection cannot run in parallel (If an attempt is active, I put everything in return, for other connections until the first one is completely completed), it only runs on a separate thread (thus relieving the main loop). I know it can cause trouble, but I don't allow this behavior. I'm not my own enemy either, I don't like unpredictable behavior :)

2, Thanks for the information, I can try this. Then I would be able to get rid of the Thread code, but unfortunately this does not solve the basic error for which I wrote the error report. :( But, the example I wrote, and the way I call the DNCClient, I don't have any errors at the moment. It's a working solution, although I don't necessarily understand why. 

Other than that, could we anytimes talk/look into the bug in my environment? I haven't written an email until now because we were at exhibitions etc. I think I'm in a different time zone than you, regardless of that, would you be part of a consultation?

megax, 2023-01-31:
I sent an email.

ssilverman, 2023-01-31:
Have you been getting my replies?

megax, 2023-01-31:
Yes, I sent a reply. For some reason it ended up in the spam folder. :(

ssilverman, 2023-02-02:
I just pushed some changes. Let me know if they fix the problem. Specifically, commit 7724fbe68ace503e2e239546c9b9c935f9cd2f1c.

ssilverman, 2023-02-07:
See: https://github.com/ssilverman/QNEthernet/issues/24#issuecomment-1421561423

ssilverman, 2023-02-08:
This commit might also help: b69228f5a72ebf960f04f4d47d620b3741c47387

megax, 2023-02-10:
I have read and tested it. I don't experience any errors when starting the router, even though I've done quite a few. So we can consider it solved, but I'm still testing it. :)

ssilverman, 2023-02-10:
I'm glad the fix works for you (so far). Thanks for doing more testing.

megax, 2023-02-13:
I tested it, in a few days, I think the error has been solved. If it appears by accident, I'll let you know.  In the meantime, I'll close the ticket. Thank you very much for the correction :)

closed 2023-02-13 by megax