ssilverman/QNEthernet #30: Errors on compile in a PlatformIO project

Nailcoding, 2023-03-17:
While using this library with teensy 4.1 and the MultiChatClient example works fine, using it in a platform IO project shows problems when compiling.
The first error is:
.pio\libdeps\teensy41\QNEthernet\src/lwipopts.h:179:36: error: expected identifier before numeric constant
 #define TCP_MSS                    1460

Renaming this macro and compiling again shows the same error in a different file as there also is a macro with this name:

.pio\libdeps\teensy41\QNEthernet\src/lwip/opt.h:1293:41: error: expected identifier before numeric constant
 #define TCP_MSS                         536

There are also some other errors like this one:
..\Firmware\PlatformIOScripts\packages\framework-arduinoteensy\libraries\FNET\src/stack/fnet_socket.h:631:1: error: expected declaration before '}' token

The part of fnet_socket.h at line 631 doesn't seem to have an error:

    IPV6_JOIN_GROUP,        /**< @brief (RFC3493) Join a multicast group on a specified local interface.@n
                            *  It is valid only for the SOCK_DGRAM (UDP) sockets.*/
    IPV6_LEAVE_GROUP        /**< @brief (RFC3493)  Leave a multicast group on a specified interface.@n
                            *  It is valid only for the SOCK_DGRAM (UDP) sockets.*/
***this is line 631:***  } fnet_socket_options_t;

My platformio.ini:
[platformio]
description = RCM
default_envs = teensy41
core_dir = ..\PlatformIOScripts

[env:teensy41]
platform = teensy
board = teensy41
framework = arduino
lib_deps = ssilverman/QNEthernet@^0.18.0

Nailcoding, 2023-03-17:
Seems it just interfered with another library (DHCP) that was included.

closed 2023-03-17 by Nailcoding

ssilverman, 2023-03-17:
I'm glad you got it sorted out. Yes, FNET is referenced by the NativeEthernet library, but isn't used by _QNEthernet_. _QNEthernet_ uses lwIP under the covers, and that's included with the library.

Note that _QNEthernet_ includes a DHCP client. (See `Ethernet.begin()` and associated examples for both blocking and non-blocking approaches.) Do you, by any chance, need to serve DHCP?