ssilverman/QNEthernet #32: Send/receive from Teensy Audio objects

hatchjaw, 2023-03-21:
Hi,

I don't know whether this constitutes a bug, or just a limitation of the library, but I've found that QNEthernet (specifically for UDP multicast transmission) doesn't work within the `update` method of an `AudioStream` descendant.

Receiving packets works for up to several thousand parses, but eventually Teensy hangs. My ethernet shield's LED continues to flicker, acknowledging incoming data, but I see no more serial activity. Attempting to send from `update()` causes Teensy to hang immediately.

We discussed a related project on the PJRC forum last year, and you [mentioned](https://forum.pjrc.com/threads/71444-Teensy-4-1-JackTrip-client-sync-issues?p=315432&viewfull=1#post315432) not being sure how QNEthernet would handle being run from an ISR, which I now understand is what I'm trying to do. 

In any case, I created a gist to illustrate the issues I've encountered: https://gist.github.com/hatchjaw/689a3833af2e79f779bd17deb0805e72

For receving packets at least, the [AsyncUDP_Teensy41](https://github.com/khoih-prog/AsyncUDP_Teensy41) library, which uses QNEthernet functions, does appear to work reliably -- though since this library operates asynchronously (or claims to), I guess it's not running on the ISR. In any case, it assigns a callback to `udp_recv()`, which in turn triggers a user-specifiable handler for each incoming packet -- very useful. Sending from the audio ISR doesn't work, however. NativeEthernet can send and receive from `update()`, but doesn't offer the advantages of lwip.

Are there insurmountable obstactles preventing QNEthernet working within an Audio Library object in this way?

ssilverman, 2023-03-21:
Which version of the library are you using? I recently updated to v0.19.0, and that fixed a few things from v0.18.0 that was causing some hangs. This might be the actual cause of the problem. Let me know if it fixes things.

Another question for curiosity: Are you using a queue size when creating the UDP packet? For example (queue size of 64):
```c++
EthernetUDP udp(64);
```
vs.
```c++
EthernetUDP udp;
```

You don't specifically need to use a queue, it's just handy when there's bursts of packets coming in all at once and there's not enough time to process them before the next arrives. (The queue size can be tuned and doesn't have to be 64.) In my experience, without that queue, fast data I was receiving and processing was getting lost.

That "async" library you mention (and all of the affiliated libraries) doesn't really use _QNEthernet_, nor is it really asynchronous. Instead, it's just another API on top of the lwIP stack ((lower-case EL as the first letter) it's the stack that _QNEthernet_ uses under the covers). It doesn't do anything better than my library, just a little differently. In fact, I explicitly disallow calling lwIP functions from inside an ISR (or ISR context) in my lwIP configuration (_lwipopts.h_). Additionally, its size requirements are larger because it causes your program to include not only (and unneeded) bits my library, but also its own on-top-of-lwIP library.

_QNEthernet_ also uses that callback: see `recvFunc()` in _QNEthernetUDP.cpp_ (and the associated `udp_recv()` call in `EthernetUDP::begin(localPort, reuse))`. When a packet is received, it's put into the circular queue. Then, `parsePacket()` will return a value >= 0 when there's a packet (zero-length included). The sequence of events:
1. Packet arrives and sets a flag inside the Ethernet ISR.
2. Some call to `Ethernet.loop()` notices there's a packet and puts it into the queue inside `EthernetUDP` via that `recvFunc()` callback. Note: `Ethernet.loop()` is called from `yield()`, after every main `loop()` call, from inside `delay()`, and also at various places sprinkled around the code.
3. When next you circle around to `loop()` again, something there or something it calls checks `parsePacket()`. That's when it takes one item off the queue and you can work with the data.

This is different from how the other library works in that the packet isn't processed until `parsePacket()` is called at some point. If packets come in fast enough, however, you'll find you need to create some sort of queue anyway, put packets into that queue in the "async" library's callback, and then check that queue later. In other words, when packet buffering is needed, you'll get to the same place with code that looks about the same as what I have in _QNEthernet_.

Fun fact: There's `EthernetUDP::data()` and `size()` convenience functions that let you get at the data directly without having to call any of the `read()` functions to load data into a separate buffer.

The library isn't designed to be operated in a concurrent (or asynchronous) fashion. There's locking that must be considered, and various data structures and variables that make some non-concurrent assumptions. The only way I can think of that _might_ work is if all _QNEthernet_ functions were called from the _same ISR/threading context_ and _not separate ones_.

Does this all make sense, and does it help? In short: try the latest _QNEthernet_ v0.19.0 and let me know how it works. I can't support use of "those other" "async" libraries since they don't really use _QNEthernet_ (just its included underlying lwIP code), but I will support _QNEthernet_ use and touch on lwIP use, if needed.

hatchjaw, 2023-03-21:
Thanks very much for the quick and detailed reply!

I've tried v0.18.0 and v0.19.0. I've also tried a few different values for the size of the packet queue, but hadn't gone as high as 64. Thanks  for shedding some light on how the packet queue works. 

> That "async" library you mention (and all of the affiliated libraries) doesn't really use QNEthernet...

My mistake; I saw that it was using functions from `QNEthernet/src/lwip`. As you say, it's their own API on top of that stack.

> It doesn't do anything better than my library, just a little differently. 

It (`AsyncUDP_Teensy41`) is also profoundly aesthetically-unpleasant code, and instructs the user to patch the core files for the `Stream` class. I gave it a try , and it appeared to be giving me decent results on the receive side of things at least, but ultimately I've found that it exhibits a similar issue to that I'm experiencing with QNEthernet -- it'll run for a few minutes, but eventually it stops calling its `onPacket` callback, and top-level `loop()` is no longer called either.

> QNEthernet also uses that callback...

Yeah, I figured that was where QNEthernet pushes to its queue. I take your point regarding the _async_ approach not really being asynchronous. There is something appealing, however, about being notified by the lwIP stack when a packet arrives, rather than calling `parsePacket()` at arbitrary intervals. I'm working on a system where ideally I want to work with small audio buffer sizes (<= 32) and have as little delay beween a packet arriving and it being decoded to audio and samples added to a buffer. That being said, I've tried running with Teensy's default (128), and I'm still finding that, calling `QNEthernetUDP::parsePacket()` on `AudioStream::update()`, I get a few thousand reads* but eventually things lock up. 

Adding a couple of calls to `Serial.print()` in `parsePacket()`, I see the following:

```
...
Packet received (#6198), From 192.168.10.10:14841, Length 134, Data: 
Before EthernetClass::loop
After EthernetClass::loop
Packet received (#6199), From 192.168.10.10:14841, Length 134, Data: 
Before EthernetClass::loop
After EthernetClass::loop
Packet received (#6200), From 192.168.10.10:14841, Length 134, Data: 
Before EthernetClass::loop
``` 
That's the last thing that's printed, so it looks suspiciously like whatever's going wrong is happening while the Ethernet stack is being moved along.

> The library isn't designed to be operated in a concurrent (or asynchronous) fashion...

And this is why I'm concerned that I'm trying to do something that QNEthernet isn't designed for, or doesn't support at present. I don't know enough about Teensy's underlying architecture, but when you mention _ISR/threading context_, for example, I imagine that audio and ethernet are handled from separate contexts, so it's inherently hazardous to be trying to carry out networking operations from an audio interrupt.

_That_ being said, I can receive _and send_ UDP packets quite reliably (and without lockups, albeit with quite a lot of jitter at times, and I'm working with a wired LAN) from `AudioStream::update()` using `NativeEthernet`. Have you any thoughts on why `NativeEthernet` works in this way where `QNEthernet` doesn't?

* well, anything from a few hundred to ~100,000. Either way, the reads eventually stop.

ssilverman, 2023-03-22:
> 
> > That "async" library you mention (and all of the affiliated libraries) doesn't really use QNEthernet...
> 
> My mistake; I saw that it was using functions from `QNEthernet/src/lwip`. As you say, it's their own API on top of that stack.

It's not clear because the projects' description does say "QNEthernet" in it, but it's not the whole story.

> > It doesn't do anything better than my library, just a little differently.
> 
> It (`AsyncUDP_Teensy41`) is also profoundly aesthetically-unpleasant code, and instructs the user to patch the core files for the `Stream` class.

Patching the core shouldn't be necessary. :(

> I gave it a try , and it appeared to be giving me decent results on the receive side of things at least, but ultimately I've found that it exhibits a similar issue to that I'm experiencing with [...]

I'm a little confused now. I'm unclear how your program works. What's calling what? What functions are being called from an interrupt? I don't know anything about this "AudioStream" library. I thought you said that you were calling `parsePacket()` from `AudioStream::update()`?

I don't think I can help further without seeing code because I'm hearing your description as contradictory. On one hand you're saying `parsePacket()` is being called from `AudioStream::update()`, but on the other, you're calling it from some `onPacket` callback function? Do I have that right?

To get equivalent behaviour as a callback-on-packet-receive, you could easily modify `EthernetUDP::recvFunc()` to call a function instead of adding to a queue. Then you could either derive from `EthernetUDP` and override that function, or make a new member that's a function. There's a few approaches.

> Yeah, I figured that was where QNEthernet pushes to its queue. I take your point regarding the _async_ approach not really being asynchronous. There is something appealing, however, about being [...]

I don't know what the code looks like, so can't provide further insights. :(

> And this is why I'm concerned that I'm trying to do something that QNEthernet isn't designed for, or doesn't support at present. I don't know enough about Teensy's underlying architecture, but when [...]

Code. :) I'm still feeling your description is contradictory, so I'm not understanding what's happening. Are there audio interrupts? Is that what's calling `update()`? If so, how is that interacting with lwIP?

> _That_ being said, I can receive _and send_ UDP packets quite reliably (and without lockups, albeit with quite a lot of jitter at times, and I'm working with a wired LAN) from [...]

I can't give thoughts without code. :(

ssilverman, 2023-03-22:
@hatchjaw if you'd like to join me via the link here, we could walk through it for a little bit:
https://forum.pjrc.com/threads/72224-Teensy-4-1-and-Ethernet-a-workshop-experiment

hatchjaw, 2023-03-22:
That's a great idea, and I'd really appreciate the walkthrough. I'm in Europe (UTC+1), but I've subscribed to that PJRC thread, so if you host another one (and can give a little notice) I'd be very happy to join.

Regarding your previous comments...

Here's the official info on Teensy Audio object (`AudioStream`) development: https://www.pjrc.com/teensy/td_libs_AudioNewObjects.html

And here's the `update()` method in the gist I posted above https://gist.github.com/hatchjaw/689a3833af2e79f779bd17deb0805e72#file-multicastnode-cpp-L28

```c++
void MulticastNode::update(void) {
    if (rx) receive();
    audioOut();
    if (tx) send();
}
```

And `MulticastNode::receive()`:

```c++
void MulticastNode::receive() {
    if (!ready) return;

    int size;
    while ((size = udp.parsePacket()) > 0) {
        udp.read(buffer, size);
        ++received;

        Serial.printf("Packet received (#%d), From ", received);
        Serial.print(udp.remoteIP());
        Serial.printf(":%d, Length %d, Data: ", udp.remotePort(), size);
//        Serial.write(buffer, size);
        Serial.println();
    }
}
```

I have an application running on my computer that's UDP multicasting packets of audio data at ~726 µs intervals (32-frames per buffer). (Note that I've also tried with larger buffer sizes, e.g. Teensy's default 128 frames.)

> I don't think I can help further without seeing code because I'm hearing your description as contradictory. On one hand you're saying parsePacket() is being called from AudioStream::update(), but on the other, you're calling it from some onPacket callback function? Do I have that right?

Apologies for the confusion; I was trying to illustrate that, using `AsyncUDP_Teensy41`, and its `onPacket` callback, I was seeing up to several minutes of transmission, whereas with QNEthernet I receive an average of a few thousand packets (so a few seconds) before Teensy locks up. Suffice it to say, I'm hesitant to use a library that insists on patching Teensy's core files; let's forget I mentioned it!

> To get equivalent behaviour as a callback-on-packet-receive, you could easily modify EthernetUDP::recvFunc() to call a function instead of adding to a queue. Then you could either derive from EthernetUDP and override that function, or make a new member that's a function. There's a few approaches.

I think `recvFunc` would have to be made protected for this to work, but yeah, that's an option -- thanks for the suggestion.

If you have time to consult the gist I created and let me know if you have any thoughts about calling `EthernetUDP::read()` and `EthernetUDP::send()` from (what I understand to be) Teensy's audio ISR, I'd be very grateful.

ssilverman, 2023-03-22:
You around? I'm there.

hatchjaw, 2023-03-22:
Forgive me; just on my way home. Any chance of going through it at 12 UTC-8?

hatchjaw, 2023-03-22:
Perfect, will do 👍

hatchjaw, 2023-03-22:
Hi Shawn; I'm available for the next couple of hours if you have time.
Tommy

ssilverman, 2023-03-22:
Ok, I'm there.

hatchjaw, 2023-03-23:
For anyone who happens upon this issue: making calls to QNEthernet from the audio ISR is not advised!

closed 2023-03-23 by hatchjaw