ssilverman/QNEthernet #33: RawFrameMonitor example seems to be missing something...

Pragmataraxia, 2023-04-06:
Would it not make more sense to put `#define QNETHERNET_ENABLE_RAW_FRAME_SUPPORT` at the top of the file rather than saying to do so in the header?

Regardless, after defining this macro, the linker still can't find the implementation for EthernetFrameClass::parseFrame().  Maybe some cpp file didn't get checked in?

ssilverman, 2023-04-06:
Thank you for your question.

That configuration macro must be defined when compiling all the files. In general, all the configuration macros need to be set at the project level because they usually affect more than one file. It's a lot easier to do if you don't use the Arduino IDE. For example, just add to `build_flags` in _platformio.ini_ if you're using PlatformIO.

I just did a search for how to do this with the Arduino IDE and came up with this:
* https://forum.arduino.cc/t/arduino-ide-where-can-i-pass-defines-to-the-compiler/680845

I know it's not the most convenient answer, but the Arduino IDE, while simple, is a little more difficult to use for more complex projects.

One of the reasons I decided to exclude the option by default is that it adds code size to projects that don't use it, and I'm assuming the majority of projects don't.

ssilverman, 2023-04-06:
Here's another link about modifying _platform.txt_ or _platform.local.txt_:
https://forum.arduino.cc/t/additional-compiler-options/631297

ssilverman, 2023-04-06:
Okay, I figured it out. Teensyduino doesn't support those "extra_flags" options in its _platform.txt_ file. So I did this:
1. Modified it to include those "extra_flags" options,
2. Added a _platform.local.txt_ with the above "compiler.cpp.extra_flags" suggestion (`compiler.cpp.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT`), and
3. Restarted the Arduino IDE (I'm using v2.0.4), and then the compilation worked.

On my system, _platform.txt_ can be found here:
_~/Library/Arduino15/packages/teensy/hardware/avr/0.59.1_
(or whichever version you have installed)

I attached my modified _platform.txt_ to this comment. I also attached the diff.
[diff.txt](https://github.com/ssilverman/QNEthernet/files/11175001/diff.txt)
[platform.txt](https://github.com/ssilverman/QNEthernet/files/11175002/platform.txt)

Another reference link:
https://arduino.github.io/arduino-cli/latest/platform-specification/

ssilverman, 2023-04-06:
I just made a post over in the Teensy forum to request a fix:
https://forum.pjrc.com/threads/72556-Request-for-Arduino-IDE-quot-extra_flags-quot-support

Pragmataraxia, 2023-04-06:
Wow... I left to get my kids and make dinner, and you've already advanced the state of Teensy development.  I hope the people around you appreciate you nearly as much as you deserve.
Cheers Man!

ssilverman, 2023-04-06:
I also updated the _RawFrameMonitor_ example commants:

```c++
// In order to use this example, define the QNETHERNET_ENABLE_RAW_FRAME_SUPPORT
// macro in the project build options.
```

ssilverman, 2023-04-07:
Turns out you need both these lines, per my setup:
```
compiler.cpp.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT
compiler.c.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT
```

Pragmataraxia, 2023-04-07:
You were right though; it was simpler in PlatformIO.  I also appreciate that it's more apparent what's going on in PlatformIO than in the Arduino IDE.  For example, I hadn't realized that the "libraries" were actually just source code.  If the Arduino IDE had bothered to mention that it was compiling QNEthernetFrame.cpp, it would have been a powerful clue that simply defining the macro before the include statement was insufficient.

ssilverman, 2023-04-07:
It wouldn’t have been sufficient because more than one file looks at the option. You’d have to set it in all those places, and that’s subject to change.

ssilverman, 2023-04-10:
@Pragmataraxia Would you consider this issue closed?

Pragmataraxia, 2023-04-10:
I sure would.  Thanks!

closed 2023-04-10 by Pragmataraxia

ssilverman, 2024-02-08:
I added a _qnethernet_opts.h_ file where you can configure the QNETHERNET_* options instead of the platform.local.txt approach.