ssilverman/QNEthernet #36: src/lwip/tcp.c causes an Access Violation.

sammburr, 2023-05-04:
The line in question the crash report is pointing to is line 1553:
`u8_t refused_flags = pcb->refused_data->flags;`

The problem is reproducible when firing the .writeFully() command multiple times in quick succession.
*Currently working on a minimal viable program to reproduce this.*

ssilverman, 2023-05-05:
I think something else is happening. `tcp_process_refused_data()`, the function where that line lives, is only called if `pcb->refused_data` is not null. I’ll look more deeply at `writeFully()` and what it does and what its calls do.

ssilverman, 2023-05-16:
@sammburr Any updates? Were you able to create a small demo?

ssilverman, 2023-06-04:
Closing. Please reopen if this is still an issue.

closed 2023-06-04 by ssilverman