ssilverman/QNEthernet #37: A couple of questions

megax, 2023-05-14:
Hi,

Clock rate question: 
I experienced a strange thing. If I reduce the clock signal from 600mhz to a lower one, e.g. 528mhz, sometimes anomalies occur in the Ethernet connection. Specifically, I cannot access the running Tcp Server from the outside. No problem at 600mhz. Currently, I can't provide sample code. Due to high heat production, I would reduce the clock signal. Do you have any idea what could be causing this?

TCP buffer question:
My other question is, is it possible that if the buffer is full, all my connections will be disconnected/not sending data? For example, one of the 3 tcp connections is broken. But the system doesn't know about it yet, max 60s later, when I check the lack of an incoming ping (there could be such a disconnection from the server side). The code continues to write and the buffer is loaded (at least it seems so). Because of this, my other connections are also broken. This is how it really happens, am I right? Is there any way I can change this? Do tcp connections share a common buffer, and if so, is it possible to set the buffer size of individual tcp connections? Thus avoiding the experienced phenomenon. (I see this phenomenon several times at 528 mhz, although I don't know if there is a connection between the two. However, I have also seen it at 600mhz. And I don't know if I'm screwing something up, or if it's actually a mistake, at least for me.)

I hope I was able to express myself clearly. Thank you in advance for your reply.

Tested: Teensy core 1.57

ssilverman, 2023-05-16:
I'm going to have a look at the different clock speeds when used with the demo programs...

ssilverman, 2023-05-16:
I just tested two TCP things at 528MHz:
1. Some custom HTTP server stuff I use for commercial projects, and
2. The IPerfServer example.

Both seem to work fine. Are you using TeensyThreads, or doing anything with QNEthernet asynchronously or from an interrupt context?

ssilverman, 2023-05-16:
What happens when you try Teensyduino 1.58?

megax, 2023-05-16:
Hi!

Thank you for your answers. I am using QNEthernet async mode. This method works perfectly with the code I emailed you. Thanks for this part. The strange thing is, there is actually no problem with EthernetClient connections. Only the EthernetServer crashes. Plus info, the microcontroller operates between 60-80 Celsius (140-176 Fahrenheit) on average. (If that matters)

I'm using your code, with a crc32 revert and turning off a couple of warnings.
https://github.com/The-Megax/QNEthernet

I will try with core 1.58. Until now, I didn't want it, I waited with it, because the new version takes an extra 20kb from RAM1. But, I'll look.
I will also update to 0.21.0. I see the final code is ready.

Thanks, I'll post it in another issue.

megax, 2023-05-17:
1. Async code
2. Speed is very important. Vincenty code calculations run into the code, plus other crc checks, and I want to minimize unnecessary time in the code.
3. PlatformIO. I haven't tried it yet, but speed is still important, and I should reduce it by a bit. 130kb of free ram1 now.

```
AsyncTCPClient::AsyncTCPClient() {
    _checkConnectedTime = millis();
    _connTimeout = 2000;
    _dnsTimeout = 1500;
}

AsyncTCPClient::~AsyncTCPClient() {
    // none
}

// Call this out of loop().
void AsyncTCPClient::loop() {
    if(_connecting && millis() - _checkConnectedTime >= _connTimeout) {
        _connecting = false;

        if(!connected()) {
            _callback(this, static_cast<int>(ConnectReturns::TIMED_OUT));
        }
        else {
            _callback(this, static_cast<int>(ConnectReturns::SUCCESS));
        }
    }
}

// Asynchronous version of connect()
bool AsyncTCPClient::asyncConnect(IPAddress ipAddr, uint16_t port, AsyncTCPClientCallback callback, network_interface_t intf /* = 0 */) {
    _callback = callback;

    int result = connectNoWait(ipAddr, port);
    if(result != static_cast<int>(ConnectReturns::SUCCESS)) {
        setConnecting(false);
        callback(this, result);
        return false;
    }

    setConnecting(true);
    return true;
}

// Asynchronous version of connect()
bool AsyncTCPClient::asyncConnect(const char* host, uint16_t port, AsyncTCPClientCallback callback, network_interface_t intf /* = 0 */) {
    _callback = callback;

    if(!DNSClient::getHostByName(host, [this, port, &callback](const ip_addr_t *foundIP) {
        if(foundIP != nullptr) {
            IPAddress l_ip = ip_addr_get_ip4_uint32(foundIP);
            int result = connectNoWait(l_ip, port);

            if(result != static_cast<int>(ConnectReturns::SUCCESS)) {
                setConnecting(true);
                callback(this, result);
            }
        }
        else {
            setConnecting(false);
            callback(this, static_cast<int>(ConnectReturns::INVALID_SERVER));
        }
    }, _dnsTimeout)) {
        setConnecting(false);
        _callback(this, static_cast<int>(ConnectReturns::INVALID_SERVER));
        return false;
    }

    setConnecting(true);
    return true;
}

void AsyncTCPClient::setConnecting(bool state) {
    _connecting = state;

    if(_connecting)
        _checkConnectedTime = millis();
}

void AsyncTCPClient::setConnectionTimeout(uint16_t timeout) {
    _connTimeout = timeout;
}
```

ssilverman, 2023-05-17:
That library doesn’t really use _QNEthernet_, with maybe the exception of initializing the PHY. I’d suggest talking with the author. It’s not really related to my library; it simply uses the lwIP stack I include.

See also this, as a side reference regarding speed:
https://github.com/ssilverman/QNEthernet/issues/39#issuecomment-1550481187

For more PlatformIO build options, see here (if you haven’t already):
https://docs.platformio.org/platforms/teensy.html
(TEENSY_OPT_SMALLEST_CODE define)

ssilverman, 2023-05-17:
Regarding the CRC-32 function, it’s only used when joining or leaving multicast groups. I’m curious, are you using it for something else?

megax, 2023-05-17:
I do not understand. I sent you a code for it a few months ago to see if it is good for dns connection. I should avoid using delay(). I totally use your library. This is now async code. But it's not the other library you think is bad. That's shit. I know it too.
I wrote this class for your tcp connection. This is a unique code. :( I only need it because of dns. Should I find the email that helped me solve this?

ssilverman, 2023-05-17:
> I do not understand. I sent you a code for it a few months ago to see if it is good for dns connection. I should avoid using delay(). I totally use your library. This is now async code. But it's not [...]

My mistake. I spoke too soon and now see it’s your code.

megax, 2023-05-17:
crc32: Yes, I use it, gps nmea format check.
Vincenty: https://en.wikipedia.org/wiki/Vincenty%27s_formulae

megax, 2023-05-17:
https://docs.platformio.org/en/latest/platforms/teensy.html
Thanks, I'll try that :)

ssilverman, 2023-05-17:
So what is the specific issue you’re seeing now, just that it doesn’t work properly at a lower frequency?

Something I just noticed: you’re performing a network function (`connectNoWait()`) inside of a callback (for DNS lookup). Try not doing that. For example, set a flag and then try the connect elsewhere. See if that helps.

megax, 2023-05-17:
Oops, thanks, I'll look into that. But unfortunately the EthernetClient is not the problem. EthernetServer bad. And I don't know why that is. And yes, it's only < 600mhz bad. In normal, 600mhz mode, absolutely fine.

ssilverman, 2023-05-17:
What specifically do you mean by “bad”?

Another question: are you 100% certain nothing else networking-related is being done asynchronously, or done from inside a network callback?

megax, 2023-05-17:
No data transfer, I can't connect to it.
It was never a problem. In fact, I also looked at 600mhz. I can't even reproduce it there. It also appears randomly on 518mhz. But it doesn't want it at 600mhz. Not in any way. That's why I'm at a loss.

I hope it doesn't happen. No threading into the code. And as long as there is no callback, the code does nothing else.
I tested a lot.

megax, 2023-05-18:
Fixed code:

```
AsyncTCPClient::AsyncTCPClient() {
    _checkConnectedTime = millis();
    _connTimeout = 2000;
    _dnsTimeout = 1500;
    setConnecting(-1);
}

AsyncTCPClient::~AsyncTCPClient() {
    // none
}

// Call this out of loop().
void AsyncTCPClient::loop() {
    if(_connecting == 2 && millis() - _checkConnectedTime >= _dnsTimeout) {
        setConnecting(-1);
        _callback(this, static_cast<int>(ConnectReturns::INVALID_SERVER));
    }
    else if(_connecting == 3) {
        int result = connectNoWait(_found_ip, _port);

        if(result != static_cast<int>(ConnectReturns::SUCCESS)) {
            setConnecting(-1);
            _callback(this, result);
        }
        else {
            setConnecting(1);
        }
    }
    else if(_connecting == 1 && millis() - _checkConnectedTime >= _connTimeout) {
        setConnecting(-1);

        if(!connected()) {
            _callback(this, static_cast<int>(ConnectReturns::TIMED_OUT));
        }
        else {
            _callback(this, static_cast<int>(ConnectReturns::SUCCESS));
        }
    }
}

// Asynchronous version of connect()
bool AsyncTCPClient::asyncConnect(IPAddress ipAddr, uint16_t port, AsyncTCPClientCallback callback, network_interface_t intf /* = 0 */) {
    _callback = callback;

    int result = connectNoWait(ipAddr, port);
    if(result != static_cast<int>(ConnectReturns::SUCCESS)) {
        setConnecting(-1);
        _callback(this, result);
        return false;
    }

    setConnecting(1);
    return true;
}

// Asynchronous version of connect()
bool AsyncTCPClient::asyncConnect(const char* host, uint16_t port, AsyncTCPClientCallback callback, network_interface_t intf /* = 0 */) {
    _callback = callback;
    setConnecting(2);

    if(!DNSClient::getHostByName(host, [this, port, &callback](const ip_addr_t *foundIP) {
        if(foundIP != nullptr) {
            IPAddress l_ip = ip_addr_get_ip4_uint32(foundIP);
            this->_found_ip = l_ip;
            this->_port = port;
            setConnecting(3);
        }
        else {
            if(_connecting == 2) {
                setConnecting(-1);
                callback(this, static_cast<int>(ConnectReturns::INVALID_SERVER));
            }
        }
    }, _dnsTimeout)) {
        setConnecting(-1);
        _callback(this, static_cast<int>(ConnectReturns::INVALID_SERVER));
        return false;
    }

    return true;
}

void AsyncTCPClient::setConnecting(int state) {
    _connecting = state;

    if(_connecting > 0)
        _checkConnectedTime = millis();
}

void AsyncTCPClient::setConnectionTimeout(uint16_t timeout) {
    _connTimeout = timeout;
}
```

ssilverman, 2023-05-18:
So now that you’re not doing network things inside the callbacks, does everything work?

megax, 2023-05-29:
I tested it. It looks good. Thanks for the help. If there are still issues, I will reopen the issue.

closed 2023-05-29 by megax