ssilverman/QNEthernet #38: TCP buffer question

megax, 2023-05-16:
Hi

My other question is, is it possible that if the buffer is full, all my connections will be disconnected/not sending data? For example, one of the 3 tcp connections is broken. But the system doesn't know about it yet, max 60s later, when I check the lack of an incoming ping (there could be such a disconnection from the server side). The code continues to write and the buffer is loaded (at least it seems so). Because of this, my other connections are also broken. This is how it really happens, am I right? Is there any way I can change this? Do tcp connections share a common buffer, and if so, is it possible to set the buffer size of individual tcp connections? Thus avoiding the experienced phenomenon.

I hope I was able to express myself clearly. Thank you in advance for your reply.

ssilverman, 2023-05-16:
Off the top of my head, is this relevant for you?
https://github.com/ssilverman/QNEthernet/blob/e78486d94d70da1b5c046aa2f2ea07c56455a62a/README.md#on-connections-that-hang-around-after-cable-disconnect

megax, 2023-05-17:
Actually, I check it. If it's because of write, betellika buffer and _client.write(msg, 1152); it does not return the size, I record it as an error. And I disconnect after max 2-3 seconds. This is EthernetServer monitoring. The EthernetClient only looks for incoming ping, but if there is none, I abort the connection for 60 seconds. This is also true for EthernetServer. There is no Ethernet cable removal, only on the opposite side, e.g. turning off Wi-Fi on the tablet. And that causes a "break". 

I used particle photon before. There is also lwip, and each tcp connection there has its own buffer. One does not affect the other. The pre-defined buffer is not common, but is dynamically reserved when needed. My question is, is there a buffer here that each TCP connection writes to, or can it be reduced/increased separately? It's okay to resend if you have one, just don't interfere with each other. So that the other's buffer is also full. When with that relationship, there is really no problem, only the other one was broken for some reason.
I increased the buffer from 24000 to 48000. But it didn't necessarily solve it.

ssilverman, 2023-05-17:
I need some time to look at this some more.

ssilverman, 2023-05-17:
Have you considered that `_client.write(msg, 1152)` might not return 1152? If the TCP pipeline is full, it might return a smaller value. See also `writeFully()` and this: https://github.com/ssilverman/QNEthernet/blob/e78486d94d70da1b5c046aa2f2ea07c56455a62a/README.md#how-to-write-data-to-connections

TCP connections depend on a lot of factors, including window size, MTU, MSS, what the remote side is doing, the pipe to the remote side, etc., etc. I’d go so far as to say that increasing MEM_SIZE might not improve behaviour. Also, TCP connections in lwIP are fairly independent; they should behave the similar to what’s in the Particle Photon.

ssilverman, 2023-05-17:
Alternatively, you could keep alternating between “write as much as will fit”, then do other things, then repeat.

megax, 2023-05-18:
Yes, as it is written, when it is full, it will not be 1152. I'm looking into this. If you don't deviate from this behavior for 3 seconds, I will assume that the connection has been lost. Then I destroy the connection with the abort() function. Unfortunately, this is not quite right with the client code. The mobile internet can fluctuate on the fields where we use the code. I have used it in the same way, until now, as a particle. It worked great there with this method, for years. It is interesting because it can do this behavior if it should actually work relatively similarly.

ssilverman, 2023-05-18:
I’m curious what happens if you use `close()` instead of `abort()`.

megax, 2023-05-19:
Thank you very much. I'll give it a try, next week I'll be able to test it longer. I'll let you know as soon as I know something.

megax, 2023-05-29:
I tested it. It looks good. Thanks for the help. If there are still issues, I will reopen the issue.

closed 2023-05-29 by megax