ssilverman/QNEthernet #39: Highly random and slow data output with large transfers over TCP (using TeensyThreads)

strud, 2023-05-16:
Hi Shawn

I know you state that the library is not shown to be 'solid' with TeensyThreads, however my project is not going to work without some form of threading architecture.

The problem I'm having is that if I am sending data over the a tcp connection that is small, it all sends immediately and no issues occur at all in my code.

If the data (file) is large, then huge random pauses in transmission occur, but eventually it seems to all get across. Pauses can be a few seconds to many minutes long.

I have the sever and client code within the loop(), with an implementation of processClientData() in an external function called from within loop().

However, I have other functions implemented in multiple threads.

The fact that it works fine for small packets (no sure of the limit but I have tested say 500 bytes) but all goes to pot with larger packets (say many MBytes in total) does this indicate a different issue not related to TeensyThreads?

I have a UDP CLI implemetation that works fine using TeensyThreads and a UDP streaming (output) approach that also works fine.

Any suggestions on where to look would be very helpful.

Craig

ssilverman, 2023-05-16:
Does calling `flush()` help?
See here: https://github.com/ssilverman/QNEthernet/tree/e78486d94d70da1b5c046aa2f2ea07c56455a62a#write-immediacy
(and its parent section)

ssilverman, 2023-05-16:
I’ll also add: all network calls ***must*** be called from the same thread or context.

strud, 2023-05-16:
No unfortunately flush does not help, it seems to make it fall over completely...

Yes I have moved the network (TCP) calls into the loop() context. However I have UDP stuff in another thread, sounds like this might be a problem?

ssilverman, 2023-05-16:
The library doesn’t support access from multiple contexts. It breaks a bunch of internal assumptions.

strud, 2023-05-16:
Yep ok will do that.

ssilverman, 2023-05-16:
I’ll add: it isn’t the case that _QNEthernet_ isn’t “solid” with interrupts or multi-threading. It’s explicitly not supported.

strud, 2023-05-16:
Understood, sorry if that came across as a negative comment, not intended.

By context do you mean that the network variables,objects etc need to be declared within the thread also? eg server(s) or can these be global without causing an issue?

strud, 2023-05-16:
Still a WIP but I have moved everything relating to the webserver and UDP CLI into a single thread.

Performance is dramatically better with MUCH higher throughput and less hangups.

I suspect there may be problems with the serial interface in TeensyThreads also, infact I think this is documented on the github repo.

Another issue may be that I need to define some vars as volatile, as I have found the compiler to do some rather odd things for variables that are shared across contexts.

ssilverman, 2023-05-16:
> Understood, sorry if that came across as a negative comment, not intended.
> 
> By context do you mean that the network variables,objects etc need to be declared within the thread also? eg server(s) or can these be global without causing an issue?

No worries. I didn't see at as negative, just "misguided" insofar as how the library works, so I was attempting to be more clear (including for future readers, who might get the wrong impression).

Regarding your question, it's any function calls. Variables and objects are just in memory, and not necessarily related to any specific context or thread, etc.

ssilverman, 2023-05-16:
> Still a WIP but I have moved everything relating to the webserver and UDP CLI into a single thread.
>
> Performance is dramatically better with MUCH higher throughput and less hangups.
>
> I suspect there [...]

I'm glad to hear that! Yeah... I think there's an impression that threads will make things faster (in general; not sure if you made that assumption or not). A lot of things can be accomplished very quickly with a single-threaded model. For example, that "flag" approach I mention above, where one context communicates to another via some "this-thing-is-ready" flag, and then the main loop checks those.

Regarding volatiles, my mental model is that it's a compile-time optimization that guarantees that access (read or write) from more than one "context" will force a reload/re-store and not rely on any cached values. Note that this is orthogonal to atomicity. "Volatile" doesn't necessarily imply "atomic". If more than one thread or "context" or ISR routines, etc, need to access a variable, then that variable needs to be marked as "volatile". (Note: There's some extra things involved with marking structs or classes as volatile.)

strud, 2023-05-16:
Yes I have some experience using volatiles when implementing a circular buffer for an ISR to feed while a loop in a thread chases the tail and streams the data byte by byte out a UDP socket.
Once I get the webserver working more reliably, I intend to change from the UDP socket stream to a Web socket stream, or at least make that an option.
Unfortunately I have to 'work' now but will be back onto this tonight.

ssilverman, 2023-05-17:
Thanks for the updates. I'm glad you're making progress. :)

strud, 2023-05-17:
So I completely removed all use of TeensyThreads and put everything into a monstrous single loop().

The webserver implementation used to hang up and perform badly with any data streams of significance. Now it is downloading files from an SD card at about 8MBytes per sec without a hitch.......

I will re-introduce TeensyThreads for non-network stuff and see what happens.

closed 2023-05-17 by strud

ssilverman, 2023-05-30:
@strud I'm curious if you re-introducted TeensyThreads for non-network stuff, and if so, what was the result?

strud, 2023-05-30:
@ssilverman no I have not as yet. When I get time will try that and report back. However worth noting that a big part of my issue was a gumby mistake of throughput calculation ie in my streaming application I was simply trying to push too much down the ethernet pipe and everything backing up.