ssilverman/QNEthernet #43: how to receive raw ethernet frames outside loop() ?

Linshiyuan, 2023-06-06:
Hi Shawn,

I am tring to receive some raw frames outside loop() in the timer interrput callback function.  The RawFrameMonitor example can receive the raw frames, but when I copy the code outsiede the loop(), it can not receive any frames. 

Is there some extra function need to be used ?  

Any suggestions would be very helpful.

ssilverman, 2023-06-06:
Asynchronous and concurrent use of the library is not supported.

See: https://github.com/ssilverman/QNEthernet/blob/007bca607f09c8c30cb335f47a8a89ae2c7d6f3a/README.md#asynchronous-use-is-not-supported

Linshiyuan, 2023-06-06:
Ok, maybe I should look at lwip and find some way.

ssilverman, 2023-06-06:
May I suggest having an event loop that consults for the last received packet (or last in the queue)? I find the performance is very good; I use this approach in most of my projects these days. Additionally, if you've chosen to queue packets (via the `EthernetFrame` or `EthernetUDP` API), you'll find that your code will tend towards something similar to what's already in the _QNEthernet_ code.

See also:
* https://forum.pjrc.com/threads/72853-Teensy-4-1-Ethernet-OctoWS2811-Art-Net-s-ACN?p=326242&viewfull=1#post326242 (the approach I take)
* https://github.com/ssilverman/QNEthernet/issues/39#issuecomment-1550481187 (a post in an issue thread)
* https://github.com/ssilverman/QNEthernet/issues/39#issuecomment-1551014434 (different post in the same thread)

Linshiyuan, 2023-06-06:
Thanks for your suggestion. 

My tast is to do send and receve frames at a vrey high accurate fixed time interval. (the EtherCAT slave device needs).  If I do the commnication in loop() funcion, the other peripherals(usb, uart..) can easely breaking the fixed time interval and the slave device get a synconization error.  Currently I use a w5500 with teensy 4.0 to run it in a timer callback. But this way is a little bit slow(200us a tx, rx cycle) and unstable.

I will try to use event loop and do some test, may be the on board eth prot is way more faster.

Linshiyuan, 2023-09-13:
Hi,
I think I am closed to my goal now, but still need some help about using the code to receive raw frame.

I use the **enet_output_frame** to send raw frame and works fine,  then I try to write a **enet_input_frame**   function in the lwip_t41.c file based on the **enet_proc_input** funtion (I guess it is doing the receive things), but my input function is return 0 all the time.
```C
// lwip_t41.c
size_t enet_input_frame(const uint8_t *frame) {
  size_t rxLen = 0;
  if (atomic_flag_test_and_set(&s_rxNotAvail)) {
    return rxLen;
  }

  for (int i = RX_SIZE*2; --i >= 0; ) {
    // Get the next chunk of input data
    volatile enetbufferdesc_t *p_bd = rxbd_next();
    if (p_bd == NULL) {
      break;
    }
    struct pbuf *p = t41_low_level_input(p_bd);
    if (p != NULL) {  // Happens on frame error or pbuf allocation error
      // Process one chunk of input data
      rxLen = p->len;
      
      memcpy(frame,p->payload,rxLen);
    
      pbuf_free(p);
    }
  }
  return rxLen;
}

bool enet_init_raw() {
 
 t41_netif_init(&s_t41_netif);

  return true;
}

// main.cpp

#include <Arduino.h>
#include "lwip_driver.h"
IntervalTimer myTimer;

int txFlag = 0;

uint8_t testData2[] = {
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x88, 0xa4, 0x0e, 0x10,
    0x07, 0x05, 0x00, 0x00, 0x30, 0x01, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
uint8_t rxBuffer[1024];

void printRX()
{
  enet_output_frame(testData2, sizeof(testData2));

  size_t rxLen = enet_input_frame(rxBuffer);
  Serial.printf("rxLen %d", rxLen);
  if (rxLen > 0 && rxLen < 1000)
  {

    for (size_t i = 0; i < rxLen; i++)
    {
      Serial.printf("%2x ", rxBuffer[i]);
    }
  }
  Serial.printf("\n");
}

void setup()
{
  Serial.begin(115200);
  enet_init_raw();
  myTimer.begin(printRX, 1000 * 1000); 
}

void loop()
{

  if (txFlag)
  {
    txFlag = 0;
    Serial.printf("send \n");
  }

  delay(200);
}

```

ssilverman, 2023-09-13:
Just curious, why not use the `EthernetFrame` API? It does whatever you’d do in C — it wraps the C calls in a way that you’d probably do yourself. Its use is similar to that of `EthernetUDP`.

ssilverman, 2023-09-13:
I won’t support concurrent/ISR-based use of the API. My suggestion is to initialize the Ethernet stack like _lwip_t41.c_ does and then use the lwIP API directly. At this point, you can configure the stack in the way you want, but I wrote everything, including the C code, with the assumption that there’s no concurrency.

Linshiyuan, 2023-09-13:
myTimer will be set to 250us or 125us  at real work situation. Here set to 1s is just for testing.

I used to do raw frame tx and rx in **loop**() with w5500 chip , it works OK when there is no other task(uart, usb serial). If there is other task blocking the loop or time varying, the jitter is huge(>50us) and all the ethercat slave device will be in error state. So I need to use a timer to trigger the tx and rx process.

ssilverman, 2023-09-13:
Three questions:
1. Is it possible to write the non-Ethernet code in a non-blocking way?
2. Why do you care about Ethernet jitter? I wouldn’t use it for precise timing unless using PTP or something.
3. Why do you need to poll Ethernet precisely?

ssilverman, 2023-09-13:
I just read a little about EtherCAT. Does it need to run under an RTOS with precise timing? If so, that’s well beyond the scope of _QNEthernet_.

If you want to chat about some consulting separately from here, I’m open to it.

Linshiyuan, 2023-09-13:
Ethercat embeds its payload in a standard ethernet frame(raw frame). The slave devices are usually some motor drivers, they require low frame jitters. (<10us). The slaves use asic chip to give a feedback msg immediately, so on the master side need to receive just after send the frame.

RTOS is not necessary, I mamaged to run my code using teensy and  w5500 chip, but the spi speed limits the tx rx frequency. The transfer time is about 200us( tx and rx frame) and will be increasing with the frame length.

So what I am doing now is to find a better way replace the w5500 chip.

ssilverman, 2023-09-13:
EtherCAT sounds like a great use case. If I have some time, I’ll read more about it.

ssilverman, 2023-09-13:
If you want to write your own input processing, you’ll need to write your own Ethernet ISR (see enet_isr()).

ssilverman, 2023-09-17:
Did replacing enet_isr() help?

Linshiyuan, 2023-09-18:
Not yet, I am tying to understand what's going on in the lwip_t41.c,  but then just get lost.... 
enet_isr clear the "s_rxNotAvail" flag, I think this is it's job ?

ssilverman, 2023-09-18:
The way I have things set up is that an interrupt is generated when Ethernet data is received. I’ve chosen to use a non-blocking approach, where, when that interrupt is generated and the ISR is called, a flag is set. That flag is later checked from the main application.

You could take any approach, it doesn’t have to be the same as what I’ve done. There’s a few approaches you could take.

ssilverman, 2023-10-02:
Closing this question for now. Please open a new Discussion topic for further discourse.

closed 2023-10-02 by ssilverman