ssilverman/QNEthernet #48: Router AP and Wifi STA question

megax, 2023-07-16:
Hi, 

This would be an unusual question, I use Teensy a bit extreme way and with it QNEthernet. We have a project where the network is completely wifi.

Operation:
Router and AP: This is the main router, here is one Teensy that acts as a "server". It runs UDP and a TCP server. It communicates with other "client" devices using UDP. The TCP server is needed for the tablet, which is used to configure it. 
Wifi STA: Here is another router and a Teensy. This Teensy is plugged into this router. The router itself works as a Wi-Fi client. It is related to the other one written above. As I wrote, Teensy communicates with UDP.

Schematic:
Teensy "server" (UDP connect) -> router and AP -> wifi sta mode (router) -> Teensy "client" (UDP connect)

Interesting experience:
The data transfer works perfectly. Even if I turn off the main router, it recovers, except for one case.
If I go too far and then get close again, the connection with Teensy will not be restored. This is the last data: LinkOff. If I check with a command, it still says this. If I unplug the cable and plug it back in, it fixes it. This is interesting because sometimes it recovers if I go far, then come closer.
I know, I'm using it too complicated, but we have to cross 350m on a rail track in the air. Later it will be a Mesh network. Maybe that will fix this problem.
QNEthernet version: 0.22 (when I tested)

Maybe I could ask for advice, an idea on how to solve this? How to reconnect?

ssilverman, 2023-07-16:
I'm having trouble following what you're saying. What I'm hearing you say is that the link becomes unstable when the physical link goes down. Sometimes "link down" doesn't always recover. For example, see here:
https://github.com/ssilverman/QNEthernet/blob/8d7614ab64a901d9ad7d4b34d3cac41f17908be0/README.md#on-connections-that-hang-around-after-cable-disconnect

Is this relevant to you?

megax, 2023-07-17:
In this case, not quite. EthernetServer is invited, but let's assume that I haven't connected to it. It runs in the background, but the one which this phenomenon occurs was not trying to connect. Well, it's not that I don't relate to it. 

Here now there are UDP connections (due to the size of the data packets and the fast delivery, I used this, so it is not TCP, which is what I communicate with first)
I have a watcher in my code so that it does not run the given parts (may send and receive data) if it is in "link down" mode. In fact, the code does not want to move from this base, it stucks there. It keeps saying "link down".

For the explanation above. There are two Teensy's in the system. They communicate with each other by using UDP. And the two are connected to the Teensy 1-1 router via ethernet. And these routers connect to each other via WiFi. So, the physical connection is unstable (or the wifi connection) and when one of the routers loses signal and then receives it again, it gets stuck in the Teensy "link down" mode. It only gets better if I restart Teensy or unplug the ethernet cable and plug it in again. 

I don't know, is it a bit more understandable? Maybe I couldn't write it properly :(

ssilverman, 2023-07-19:
I’m so sorry, but I’m still having trouble understanding the specific issue. Is it that once the link goes down it stays down? If that’s the case, could it be that the router the Teensy is connected to is misbehaving and it’s not the Teensy that’s having problems?

megax, 2023-07-19:
Yes, it stays that way. It is possible, but if I unplug the cable that is in Teensy and plug it in again (ethernet), the connection is restored. Also in case of Teensy restart. That's actually my question. Could I somehow find out if the router is faulty or if Teensy sees it incorrectly? Is there a way to do this?

(I tried to describe a phenomenon, do you have any advice for it? I clearly use it for something that goes beyond the average use. I'm already happy that it works at all.)

megax, 2023-07-21:
OK thank you. I will try to clarify next week and possibly send a logo. But, I don't want to bore you with it. If there isn't anything that might be meritorious, as an idea. Maybe if I'm there again for a project (it's not in my country, so I have to build demo devices to be able to test it too), I'll talk to my boss and we'll discuss development costs/help if that's okay with you too. As I wrote, I don't use it for "normal" things, so I'm sure that this is partly the reason why the router doesn't respond well, or possibly teensy. As you wrote, there is less chance of this.

ssilverman, 2023-07-23:
Sure, I'm open to doing development help, but contact me privately for that.

megax, 2023-07-31:
I'm closing this topic now. Because it's just a question, and I even need to gather more information about it. If that is the case and the error is present, we will search based on my last statement, I think. Until then, I'm still learning from this way of using it. Or what I am possibly doing wrong. Thanks for the help plus the information and tips. :)

closed 2023-07-31 by megax