ssilverman/QNEthernet #51: UDP Data Access Violation error

megax, 2023-09-20:
Hello,

I have found a new error. Learning from the mistakes I made in a previous issue  (https://github.com/ssilverman/QNEthernet/issues/47#issuecomment-1637553982), I'll try to handle this one as a separate issue.
Im using an updated version of the code I sent you in the previous one. I'll send you the new version as soon as I can.
That one fixed some errors, but I'm guessing this is a completely new type, due to me using UDP in the new one.

This is the new error:

```
CrashReport:
  A problem occurred at (system time) 13:44:27
  Code was executing from address 0x1CE40
  CFSR: 82
(DACCVIOL) Data Access Violation
(MMARVALID) Accessed Address: 0x31333A
  Temperature inside the chip was 60.93 °C
  Startup CPU clock speed is 528MHz
  Reboot was caused by auto reboot after fault or bad interrupt detected
```

This is the error I see after checking the memory address, this is the path:
```
PS C:\Users\JakosaCsaba\Documents\HariTech\gitrepos\robot\.pio\build\julich3> C:\Users\JakosaCsaba\.platformio\packages\toolchain-gccarmnoneeabi-teensy\bin\arm-none-eabi-addr2line.exe -e .\firmware.elf -a 0x1CE40
0x0001ce40
(...)\QNEthernet\src/QNEthernetUDP.cpp:236
```
I redacted the folders, but, this is where it points at:
https://github.com/ssilverman/QNEthernet/blob/master/src/QNEthernetUDP.cpp#L236

This is a complete mistery to me. There are two Teensy, both with the same code installed. They are communicating with each other. I use one as "server" and the other as "client". The error comes randomly, only at the one I use as "server", and every time, i get the same path. 
I have one guess, but then it wouldn't throw/generate this error: https://github.com/tonton81/FlexCAN_T4
I had a problem in this directory before. I tried using UDP `send()` function inside FlexCAN `read()`. I told me that I am using it in interrupt mode, so I turned it off.
Can this be the problem?
TCP connections work fine (If i dont use UDP)

Can you help me what might cause this, am I missing something?

I'll send sample code as soon as possible, but I can't promise that I can reproduce the error.
(Perhaps I can send you the elf code in private, if it helps, I can't send the source)

QNEthernet version: v0.24.0
Teensy: 1.59 beta

ssilverman, 2023-09-20:
> This is the error I see after checking the memory address, this is the path:
>
> ```
> PS C:\Users\JakosaCsaba\Documents\HariTech\gitrepos\robot\.pio\build\julich3> [...]

I don't think the line number changed between versions, but the correct link to the specific line is the following. The reason is that _master_ might be different than the version you're using.

Permalink: https://github.com/ssilverman/QNEthernet/blob/5ac821f64b1e4eb0a0e1f5b26e1d6ebe78d70d1b/src/QNEthernetUDP.cpp#L236
Or this: https://github.com/ssilverman/QNEthernet/blob/v0.24.0/src/QNEthernetUDP.cpp#L236
(These are from the 0.24.0 tag and it does look like the code is the same.)

> This is a complete mistery to me. There are two Teensy, both with the same code installed. They are communicating with each other. I use one as "server" and the other as "client". The error comes [...]

Is `read()` (and in turn, UDP's `send()`) called from an ISR? If so, one technique you could use is to set a flag, and then check that flag from somewhere in the main loop. This is what I do to handle Ethernet input. See `enet_isr()` in _lwip_t41.c_. If you're not doing that, I think I would need to see some code, preferably a minimal code example that reproduces this. You could email it to me and I'll have a look, but please make it as small as you can.

It doesn't make sense that the line where the error is reported as occurring is actually where the problem is. You probably feel the same way about that line. :)
This is why I suspect interrupts.

> I'll send sample code as soon as possible, but I can't promise that I can reproduce the error. (Perhaps I can send you the elf code in private, if it helps, I can't send the source)

I don't mind taking a quick look, but please make it small. I don't have much bandwidth these days.

> QNEthernet version: v0.24.0 Teensy: 1.59 beta

Which beta version are you using? The latest (as of this writing) is 3.

megax, 2023-09-21:
Yes, sorry, I put the wrong link for that line, should have been the right. It does match, but my mistake,

This way of using ISR always felt a little like "magic" to me, if you that makes sense to you. I fear that I mess it up, thats why I tried solving it in another way. I saw somewhere in an example code or documentation (I dont remember where) `enableFIFOInterrupt();`
This should turn it on if I call it. In the examples it was called, so I dont call it, then ISR is not active. Instead the `read()` gives canbus values, but there is no ISR.
ISR: https://github.com/tonton81/FlexCAN_T4/blob/master/examples/CAN2.0_example_FIFO_with_interrupts/CAN2.0_example_FIFO_with_interrupts.ino#L16
NO ISR: https://github.com/tonton81/FlexCAN_T4/blob/master/examples/read_two_channels/read_two_channels.ino#L17

If I can get an example, I can try, or especially what You wrote with enet_isr(). Should I do it by calling that?
Sorry, this is something that I really think I'm using wrong, wherever I have to use it. This is new for me too, Im not used to handling these. I've always used default/contemporary thread-safety. This is new for me.

Teensy 1.59 beta 2.

I'll send you an example code asap. What I used to use, I'll send you it's elf file in private e-mail. I'll try to compress it. If you find that it is too big, please ignore it.
I'll try to send it soon, and of course I try to reprudce the error with that as well.

Yes, I feel the same, that the reason for the error is not where the memory points at :(

megax, 2023-09-21:
Sorry, I have misunderstood what you meant by small amount of source code.
I am behind schedule at work, some other works came up, I'll send you the sample next week as soon as can.
I want a sample that actually produces error.

ssilverman, 2023-09-21:
I’ll add: by “bandwidth” I meant “personal time” and not “internet bandwidth.” :)

megax, 2023-09-21:
Oh, alright, now I understand :)

megax, 2023-09-21:
I'm not 100% sure, but i think i found the problem. I'm still testing. If it is, I'll message you and write down what the problem was. It is likely that I made a mistake.

ssilverman, 2023-09-21:
Great! I’d love to hear what the solution was.

ssilverman, 2023-09-22:
Try using `snprintf()` instead of `sprintf()`, and also check return values and see if that helps.

In other words, here are some general tips:
1. Don’t use unbounded string functions, and
2. Always check return values. (In general, not just string functions.)

megax, 2023-09-25:
Thank you, I'll use the recommendations. Again, I'm sorry, although I tried,I should have tried more to solve it.
I'll close the issue.

Thanks in advance.

closed 2023-09-25 by megax