ssilverman/QNEthernet #54: Enabling Raw Frame Support and Promiscuous

owaits, 2023-10-31:
It took me a while to work out how you enable both raw frame support and promiscuous together as the readme only gives an example for enabling one macro. Can I suggest updating the example to include enabling two macros as bellow.

```
compiler.cpp.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT -DQNETHERNET_ENABLE_PROMISCUOUS_MODE
compiler.c.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT -DQNETHERNET_ENABLE_PROMISCUOUS_MODE
```

It all seems obvious now but many of the forum posts I read suggested quotes which lead me down the garden path. In the end I had omitted the -D from  QNETHERNET_ENABLE_PROMISCUOUS_MODE as I had not spotted in the readme that this had been appended to QNETHERNET_ENABLE_RAW_FRAME_SUPPORT.

Amazing library though, thank you so much for your work on this.

ssilverman, 2023-10-31:
Thanks for using _QNEthernet_ and for the suggestion.

I'm curious, can you point me to where you read about using quotes?

ssilverman, 2023-10-31:
Also, I'm curious: what did your non-working attempts look like? Would you mind listing them?

owaits, 2023-10-31:
The forum article that suggested quotes was this [](https://forum.arduino.cc/t/how-to-specify-multiple-extra-compiler-flags/994083)

Here are some of the variants I tried:

```
compiler.cpp.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT, QNETHERNET_ENABLE_PROMISCUOUS_MODE

compiler.cpp.extra_flags=-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT -QNETHERNET_ENABLE_PROMISCUOUS_MODE

compiler.cpp.extra_flags="-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT -QNETHERNET_ENABLE_PROMISCUOUS_MODE"

compiler.cpp.extra_flags='-DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT -QNETHERNET_ENABLE_PROMISCUOUS_MODE'

compiler.cpp.extra_flags=-QNETHERNET_ENABLE_PROMISCUOUS_MODE -DQNETHERNET_ENABLE_RAW_FRAME_SUPPORT  //Resulted in compile error

```

I have been trying to receive LLDP packets, in the end I did not need to enable it as the reason it was not working was because of the line

`if (size < EthernetFrame.minFrameLen() - 4) `

in the RawFrameMonitor example. This ignores any raw packets over about 41 bytes and the LLDP packets where 491 bytes.

ssilverman, 2023-10-31:
[The reply appears to be missing the forum link.]

That code in _RawFrameMonitor_ checks for _less than_  60 bytes (64 - 4) and prints "SHORT Frame", and then prints the whole frame if >= 60 bytes. I don't think it ignores 491-byte frames?

owaits, 2023-10-31:
I have updated my comment to include the forum link. I had over simplified the RawFrameMonitor example as I just wanted to see the raw data printed to serial. The example code as written is fine.

ssilverman, 2023-11-04:
See commit 54847045.

closed 2023-11-04 by ssilverman