ssilverman/QNEthernet #55: LTO Support

Stevenlawrencehoriba, 2023-11-09:
Any idea when LTO support will be fixed to remove these type of warnings

.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:32:7: warning: type 'struct EthernetClient' violates the C++ One Definition Rule [-Wodr]
   32 | class EthernetClient : public Client {
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.h:32:7: note: a different type is defined in another translation unit
   32 | class EthernetClient : public Client {
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:176:47: note: the first difference of corresponding definitions is field 'conn_'
  176 |   std::shared_ptr<internal::ConnectionHolder> conn_;
      |                                               ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.h:176:47: note: a field of same name but different type is defined in another translation unit
  176 |   std::shared_ptr<internal::ConnectionHolder> conn_;
      |                                               ^
C:/Users/Lawrences/.platformio/packages/toolchain-arm-cortexm-win64/arm-cortexm7f-eabi/include/c++/13.2.0/bits/shared_ptr.h:175:11: note: type 'struct shared_ptr' itself violates the C++ One Definition Rule
  175 |     class shared_ptr : public __shared_ptr<_Tp>
      |           ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetServer.h:84:18: warning: 'accept' violates the C++ One Definition Rule [-Wodr]
   84 |   EthernetClient accept() const;
      |                  ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetServer.cpp:82:16: note: return value type mismatch
   82 | EthernetClient EthernetServer::accept() const {
      |                ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.h:32:7: note: type 'struct EthernetClient' itself violates the C++ One Definition Rule
   32 | class EthernetClient : public Client {
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:32:7: note: the incompatible type is defined here
   32 | class EthernetClient : public Client {
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetServer.cpp:82:16: note: 'accept' was previously declared here
   82 | EthernetClient EthernetServer::accept() const {
      |                ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetServer.cpp:82:16: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:118:8: warning: type of 'flush' does not match original declaration [-Wlto-type-mismatch]
  118 |   void flush() final;
      |        ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:459:6: note: 'flush' was previously declared here
  459 | void EthernetClient::flush() {
      |      ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:459:6: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:116:10: warning: type of 'write' does not match original declaration [-Wlto-type-mismatch]
  116 |   size_t write(const uint8_t *buf, size_t size) final;
      |          ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:405:8: note: 'write' was previously declared here
  405 | size_t EthernetClient::write(const uint8_t *buf, size_t size) {
      |        ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:405:8: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:64:11: warning: type of 'connected' does not match original declaration [-Wlto-type-mismatch]
   64 |   uint8_t connected() final;
      |           ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:137:9: note: 'connected' was previously declared here
  137 | uint8_t EthernetClient::connected() {
      |         ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:137:9: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:120:7: warning: type of 'available' does not match original declaration [-Wlto-type-mismatch]
  120 |   int available() final;
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:483:5: note: 'available' was previously declared here
  483 | int EthernetClient::available() {
      |     ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:483:5: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:74:8: warning: type of 'stop' does not match original declaration [-Wlto-type-mismatch]
   74 |   void stop() final;
      |        ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:205:6: note: 'stop' was previously declared here
  205 | void EthernetClient::stop() {
      |      ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:205:6: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:121:7: warning: type of 'read' does not match original declaration [-Wlto-type-mismatch]
  121 |   int read() final;
      |       ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:516:5: note: 'read' was previously declared here
  516 | int EthernetClient::read() {
      |     ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:516:5: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:65:12: warning: type of '__conv_op ' does not match original declaration [-Wlto-type-mismatch]
   65 |   explicit operator bool() final;
      |            ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:155:1: note: '__conv_op ' was previously declared here
  155 | EthernetClient::operator bool() {
      | ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:155:1: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:35:3: warning: type of '__dt_comp ' does not match original declaration [-Wlto-type-mismatch]
   35 |   ~EthernetClient();
      |   ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:57:1: note: '__dt_comp ' was previously declared here
   57 | EthernetClient::~EthernetClient() {
      | ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:57:1: note: code may be misoptimized unless '-fno-strict-aliasing' is used
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src/QNEthernetClient.h:34:3: warning: type of '__ct_comp ' does not match original declaration [-Wlto-type-mismatch]
   34 |   EthernetClient();
      |   ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:50:1: note: '__ct_comp ' was previously declared here
   50 | EthernetClient::EthernetClient() : EthernetClient(nullptr) {}
      | ^
.pio\libdeps\teensy41_dbgx_exram\QNEthernet\src\QNEthernetClient.cpp:50:1: note: code may be misoptimized unless '-fno-strict-aliasing' is used
lto-wrapper.exe: warning: using serial compilation of 6 LTRANS jobs
lto-wrapper.exe: note: see the '-flto' option documentation for more information

ssilverman, 2023-11-09:
I tried building with LTO but the build doesn't fail. I tried adding the `-flto` build flag, and I also tried with setting `-DTEENSY_OPT_FASTER_LTO`. I'm using PlatformIO.

What version of Teensyduino are you using?

Stevenlawrencehoriba, 2023-11-10:
[env:teensy41_dbgx_exram]
platform = https://github.com/tsandmann/platform-teensy.git
board = teensy41
framework = arduino
lib_deps = 
	https://github.com/tsandmann/freertos-teensy.git
	https://github.com/PaulStoffregen/SPI
	https://github.com/greiman/SdFat
	https://github.com/nickgammon/Regexp
	ssilverman/QNEthernet@^0.25.0
build_Unflags = -std=gnu++17	
build_flags = -std=gnu++20 -Wformat=1 -fno-strict-aliasing -Wno-deprecated -Wno-unused-variable -Wno-unused-parameter
	-D DEFAULT_TIME -D WAIT_SERIAL -D DEBUG_IFC -D DIAG_TASKS -D QSPI_RAM -DTEENSYDMX_USE_INTERVAL_TIMER 
	-DTEENSY41_RTOS -DARDUINO_TEENSY41 -DUSB_SERIAL -DTEENSY_OPT_FASTER_LTO -DLWIP_MDNS_RESPONDER=0
monitor_speed = 115200
monitor_rts = 0

ssilverman, 2023-11-12:
I can't seem to reproduce the build error.

ssilverman, 2023-11-21:
Closing for now. Please reopen if this is still an issue for you.

closed 2023-11-21 by ssilverman

ssilverman, 2023-12-29:
I just noticed this, with the capital ‘U’: “build_Unflags”
Try with “build_unflags” instead.