ssilverman/QNEthernet #59: ReadMe file for MbedTLSDemo has no file "mbedtls/src/mbedtls/config.h"

Mike4U, 2023-12-01:
In ReadMe:
'Last, modify the mbedtls/src/mbedtls/config.h file by replacing it with the contents of examples/MbedTLSDemo/sample_mbedtls_config.h. '

Perhaps a little above that:
'Next, create an empty mbedtls.h file inside "Libraries"/mbedtls/src/.'

should be:
Next, create an empty mbedtls.h file inside "Libraries"/mbedtls/src/mbedtls/    ?

Mike4U, 2023-12-01:
Sorry, I wasn't clear.  The readme says to modify a file that does not exist at that point:  mbedtls/src/mbedtls/config.h

ssilverman, 2023-12-01:
Thanks. I’ll double-check the instruction ordering and the file names.

Mike4U, 2023-12-01:
So I made a mbedtls/src/mbedtls/config.h with the contents of examples/MbedTLSDemo/sample_mbedtls_config.h and after additionally coping  certs.h to ~/Documents/Arduino/libraries/mbedtls/src/mbedtls/ from ~/mbedtls/tests/include/test/certs.h now get error:
/Users/michaelrunyan/Documents/Arduino/libraries/QNEthernet/src/lwip/apps/altcp_tls/altcp_tls_mbedtls.c:85:10: fatal error: mbedtls/ssl_internal.h: No such file or directory
   85 | #include "mbedtls/ssl_internal.h" /* to call mbedtls_flush_output after ERR_MEM */
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

ssilverman, 2023-12-01:
Thanks for trying all that out. I went through it all myself before pushing the changes, but maybe I missed something. I’ll go through it again.

ssilverman, 2023-12-02:
See e401f996.

Mike4U, 2023-12-02:
get:
Using board 'teensy41' from platform in folder: ~/Library/Arduino15/packages/teensy/hardware/avr/0.59.3
Using core 'teensy4' from platform in folder: ~/Library/Arduino15/packages/teensy/hardware/avr/0.59.3

Using library QNEthernet at version 0.26.0-snapshot in folder: ~/Documents/Arduino/libraries/QNEthernet 
Using library Mbed TLS at version 2.28.6 in folder: ~/Documents/Arduino/libraries/mbedtls

ssilverman, 2023-12-02:
Hmm... And there's no _src/mbedtls/config.h_ file? It should have been copied from here: https://github.com/Mbed-TLS/mbedtls/blob/v2.28.6/include/mbedtls/config.h

Mike4U, 2023-12-02:
I did select all in folder, copy and paste in folder I think.

Still checking, I need coffee.

Mike4U, 2023-12-02:
I was possibly foiled by my lack of GitHub desktop skills again.  I see that the branch (development) has 27 commits.  Could that be it?

Mike4U, 2023-12-02:
Fetching the commits didn't help.
However, changing the current branch in GitHub desktop to mbedtls-2.28 brought config.h in.

For some reason there is not a 2.28.6 tag.  Can't even show tags on desktop?  I thought desktop was supposed to match the web.

Mike4U, 2023-12-02:
Got a bit further with new error:
In file included from /Users/michaelrunyan/Documents/Arduino/libraries/mbedtls/src/bignum_core.c:21:
/Users/michaelrunyan/Documents/Arduino/libraries/mbedtls/src/bn_mul.h:27:10: fatal error: mbedtls/build_info.h: No such file or directory
   27 | #include "mbedtls/build_info.h"

I'm going to delete the repository and reclone it with the v2.28.8 tag active on GitHub.com and see if that fixes it.

Mike4U, 2023-12-02:
Still can't get a tagged configuration on Github desktop.  I'll try just downloading the zip.

ssilverman, 2023-12-02:
> I was possibly foiled by my lack of GitHub desktop skills again. I see that the branch (development) has 27 commits. Could that be it?

Use either the `v2.28.6` tag for version 2.28.6 or the `mbedtls-2.28` branch for the latest development version of 2.28.x. (Note that there's also an `mbedtls-2.28.6` tag that points to the same place as the `v2.28.6` tag.)

> Fetching the commits didn't help. However, changing the current branch in GitHub desktop to mbedtls-2.28 brought config.h in.
> 
> For some reason there is not a 2.28.6 tag. Can't even show tags on desktop? I thought desktop was supposed to match the web.

It needs a leading `v`. The tag name is `v2.28.6`.

> Still can't get a tagged configuration on Github desktop. I'll try just downloading the zip.

Try the `mbedtls-2.28` branch.

Mike4U, 2023-12-02:
the mbedtls-2.28 got me to the error 19 minutes ago:  no build_info.h

ssilverman, 2023-12-02:
> the mbedtls-2.28 got me to the error 19 minutes ago: no build_info.h

_include/mbedtls/build_info.h_ (or, if you've copied the files, _src/mbedtls/build_info.h_) doesn't exist in v2.x.x. It was introduced in v3.0.0 according to mbedtls's ChangeLog. I think you have a mixture of files in there.

ssilverman, 2023-12-02:
> a bunch of files from the GitHub desktop mbedtls branch mbedtls-2.28 have #include "mbedtls/build_info.h"

I disagree. I just checked out that branch, did a global search, and found no text with "build_info.h".

ssilverman, 2023-12-02:
If you're on a Mac or Linux (not sure how to do Windows), try these commands:
1. `git clone https://github.com/Mbed-TLS/mbedtls.git`
2. `cd mbedtls`
3. `git checkout v2.28.6`
4. `grep -r "build_info.h" .`

Note that there's nothing found (except maybe in the objects cache).

To compare, check out the `development` branch (this is the default, I believe), and run the `grep` command again. You'll see that it finds many files with this string.

Mike4U, 2023-12-02:
Yaa the sucker finally compiled.  There was an issue when copying from      distro/include/* -> "Libraries"/mbedtls/src
There was already a CMakeLists.txt from the distro/library copy.  I kept both

ssilverman, 2023-12-02:
> Yaa the sucker finally compiled. There was an issue when copying from distro/include/* -> "Libraries"/mbedtls/src There was already a CMakeLists.txt from the distro/library copy. I kept both

Are you using cmake to build? If not, those files shouldn't matter.

Mike4U, 2023-12-02:
Looking at my last comment.  To be clear, it compiled with both CMakeLists.txt and CMakeLists 2.txt present.

Mike4U, 2023-12-02:
No, just for your info

ssilverman, 2023-12-02:
Well, I'm glad you got it to build. Thanks for trying out the process in the README. Other than git-specific things, let me know what difficulties and other hiccups you encounter.

Mike4U, 2023-12-02:
Still don't know why the git desktop didn't work.  It could be because I didn't clear all the files from Libraries"/mbedtls/src before I copy/pasted the files in.  I'll try again just for fun.

Mike4U, 2023-12-02:
Retried with git DeskTop first clearing all the files from Libraries"/mbedtls/src and it worked.  Guess Arduino was trying to compile any file it saw in those directories.  Well that was a painful live and learn!

Now I'll try to actually do something with it.

Thanks a ton.

ssilverman, 2023-12-02:
> Guess Arduino was trying to compile any file it saw in those directories.

It does do that. I think most IDEs that aren't using Make, or don't have some scheme where you specify specific files, will do that.

Mike4U, 2023-12-02:
thanks again

closed 2023-12-02 by Mike4U