ssilverman/QNEthernet #60: Compile errors with Teensyduino 1.59-beta4

PaulStoffregen, 2023-12-03:
Looks like some changes that went into the core library might be causing problems?

If I need to change something in the core library, hopefully we can discuss and get this fixed before 1.59 release.

For example, when compiling PixelPusherServer...

```
/tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino: In lambda function:
/tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:65:42: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   65 |     networkChanged(Ethernet.localIP() != INADDR_NONE, state);
      |                                          ^~~~~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:33,
                 from /tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:13:
/home/paul/Arduino/libraries/QNEthernet/src/util/ip_tools.h:27:6: note: candidate 1: 'bool qindesign::network::operator!=(const IPAddress&, const IPAddress&)'
   27 | bool operator!=(const IPAddress &a, const IPAddress &b);
      |      ^~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:15,
                 from /tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:13:
/home/paul/.arduino15/packages/teensy/hardware/avr/0.59.4/cores/teensy4/IPAddress.h:82:14: note: candidate 2: 'bool IPAddress::operator!=(const IPAddress&) const'
   82 |         bool operator!=(const IPAddress& addr) const {
      |              ^~~~~~~~
/tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino: In lambda function:
/tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:71:25: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   71 |     bool hasIP = (ip != INADDR_NONE);
      |                         ^~~~~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:33,
                 from /tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:13:
/home/paul/Arduino/libraries/QNEthernet/src/util/ip_tools.h:27:6: note: candidate 1: 'bool qindesign::network::operator!=(const IPAddress&, const IPAddress&)'
   27 | bool operator!=(const IPAddress &a, const IPAddress &b);
      |      ^~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:15,
                 from /tmp/.arduinoIDE-unsaved2023113-3452708-4794bl.xqh6g/PixelPusherServer/PixelPusherServer.ino:13:
/home/paul/.arduino15/packages/teensy/hardware/avr/0.59.4/cores/teensy4/IPAddress.h:82:14: note: candidate 2: 'bool IPAddress::operator!=(const IPAddress&) const'
   82 |         bool operator!=(const IPAddress& addr) const {
      |              ^~~~~~~~
/home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.cpp: In member function 'bool qindesign::network::EthernetClass::begin(const IPAddress&, const IPAddress&, const IPAddress&, const IPAddress&)':
/home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.cpp:186:11: error: ambiguous overload for 'operator!=' (operand types are 'const IPAddress' and 'const IPAddress')
  186 |   if (dns != INADDR_NONE) {
      |       ~~~ ^~ ~~~~~~~~~~~
      |       |      |
      |       |      const IPAddress
      |       const IPAddress
/home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.cpp:186:11: note: candidate: 'operator!=(uint32_t {aka long unsigned int}, uint32_t {aka long unsigned int})' (built-in)
  186 |   if (dns != INADDR_NONE) {
      |       ~~~~^~~~~~~~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:15,
                 from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.cpp:7:
/home/paul/.arduino15/packages/teensy/hardware/avr/0.59.4/cores/teensy4/IPAddress.h:82:14: note: candidate: 'bool IPAddress::operator!=(const IPAddress&) const'
   82 |         bool operator!=(const IPAddress& addr) const {
      |              ^~~~~~~~
In file included from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.h:33,
                 from /home/paul/Arduino/libraries/QNEthernet/src/QNEthernet.cpp:7:
/home/paul/Arduino/libraries/QNEthernet/src/util/ip_tools.h:27:6: note: candidate: 'bool qindesign::network::operator!=(const IPAddress&, const IPAddress&)'
   27 | bool operator!=(const IPAddress &a, const IPAddress &b);
      |      ^~~~~~~~

exit status 1

Compilation error: exit status 1
```

ssilverman, 2023-12-04:
Hi, Paul! Thanks for the report. Whenever there’s a new release, I check all the examples and run all the unit tests. I actually fixed all the issues earlier today. :)
(I just haven’t pushed yet.)

Some background on why these errors are there: The changes introduced in Teensyduino 1.59 are the right ones. I introduced some of my own IPAddress (external to the class) operators and functions, and those are the ones that conflict. They aren’t needed anymore and I’ve already added a check for the Teensyduino version with a few #ifdefs.

Thanks again for noticing.

PaulStoffregen, 2023-12-04:
Looks like I spoke too soon.  The script found compile errors with RawFrameMonitor and OSCPrinter.

The OSCPrinter just looks like a missing library.

ssilverman, 2023-12-04:
I fixed the _RandomNumbers_ compile warning; it was an incorrect `printf` conversion specifier. Pushed.

closed 2023-12-06 by ssilverman