ssilverman/QNEthernet #68: Compile Error in Arduino UI 1.8.19

shmorgan, 2024-03-19:
Compile Error in Arduino 1.8.19.

Added #include <cstdlib> to line 9 to correct.

C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\sys_arch.cpp:192:3: error: 'srand' is not a member of 'std'
   std::srand(sys_now());
   ^
C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\sys_arch.cpp:192:3: note: suggested alternative:
In file included from c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\sys\unistd.h:8:0,
                 from c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\unistd.h:4,
                 from C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\sys_arch.cpp:8:
c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\stdlib.h:154:7: note:   'srand'
 _VOID _EXFUN(srand,(unsigned __seed));
       ^
In file included from C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\security/RandomDevice.h:13:0,
                 from C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\sys_arch.cpp:23:
C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src/StaticInit.h: In instantiation of 'StaticInit<T>::StaticInit() [with T = qindesign::security::RandomDevice]':
C:\Users\Stephen Morgan\Documents\Arduino\libraries\QNEthernet\src\security/RandomDevice.h:47:1:   required from here

ssilverman, 2024-03-20:
1. Which version of Teensyduino?
2. If not a Teensy, which platform and version?
3. Which version of the library?

shmorgan, 2024-03-20:
I did a complete removal and re-install of Arduino 1.8.19 and Teensyduino 1.57 and this issue seems to be resolved.

ssilverman, 2024-03-20:
Have you tried Teensyduino 1.59? I recommend that instead.