ssilverman/QNEthernet #70: QNEthernet forr teensy 4.1 and OTHER  platforms questions

5b4wn, 2024-03-23:
Dear ssilverman

MANY THANKS for your library. I have successfully used it in the past for  hobby projects using teensy 4.1. I am now playing with RP2040 cores using W5500 chipset (eg W5500-EVB-Pico) which is very inexpensive and quite powerful. The question is would your QNEthernet library work with some mods (as an OTHER platform). The first compilation brings up WString.h error and even when changing this to <api/String.h> as suggested  https://github.com/earlephilhower/arduino-pico/discussions/1079 more errors come about eg
QNMDNS.h:49:3: error: 'String' does not name a type
   49 |   String hostname() const {

Do you have any suggestions or should I stay with the Ethernet.h? I am after websocket server functionality in my project and somehow the setup with websockets2_generic doesnt work as expected with any of the ethernet libraries.

BW
Marios

ssilverman, 2024-03-23:
I'll try to compile the _QNEthernet_ examples here for that platform. Can you tell me more about your build setup?

Note that the "other platforms" might or might not work for a given platform because of how varied Arduino-style implementations are.

5b4wn, 2024-03-23:
Hi

many thanks for trying and I understand that it may not work out.... but if it does it would be awesome...

Just using the Wiznet W5500-EVB-PICO option in arduino IDE
FYI  the W5500-EVB PICO https://www.wiznet.io/wp-content/uploads/2022/04/w5500-evb-pico-pinout.png
RP2040 is connected as follows to the W5500
MISO - 16
MOSI -19
CLK-  18
SS- 17

BW
5B4WN

ssilverman, 2024-03-23:
Can you tell me more about your build setup? Arduino IDE? PlatformIO? Which platform did you select? If PlatformIO, what does your platformio.ini file look like?

5b4wn, 2024-03-23:
Hi Shawn

using arduino IDE

[image: image.png]

from https://github.com/earlephilhower/arduino-pico

Hope this helps

BW
Marios

ssilverman, 2024-03-23:
There's no image attached. Could you attach it via this page instead of email? (https://github.com/ssilverman/QNEthernet/issues/70)

5b4wn, 2024-03-23:
[image]

ssilverman, 2024-03-23:
I tried building some examples, and none could compile. It can't find things like `IPAddress`. I'm using Arduino IDE 2.3.2.

Here were my steps:
1. Add `https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json` to  "Additional boards manager URLs" in the IDE settings.
2. Install the "Raspberry Pi Pico/RP2040" board in the Boards Manager tab.
3. Change the board to "WIZnet W5500-EVB-Pico".
4. Try to build several of the _QNEthernet_ examples.

I don't think this board setup was created properly. I looked in the project's repo and saw that there _are_ headers such as `WString.h` and `IPAddress.h`. Something with how that project is structured doesn't work.

5b4wn, 2024-03-24:
Thank you @ssilverman .

I will ask @earlephilhower author of RP2040 core libraries if he has any suggestions..

BW
@5b4wn

ssilverman, 2024-03-24:
Related: https://github.com/earlephilhower/arduino-pico/issues/2071

ssilverman, 2024-03-24:
I just tried to create a simple sketch that includes <WString.h> and <IPAddress.h> with that WIZnet board, and it compiles fine. There's something about including those headers from libraries that isn't working. Because there's already networking support there, I don't plan on spending much more time on this.

5b4wn, 2024-03-24:
Dear @ssilverman 

many thanks for looking at this! I will keep scratching my head..

Have a great Sunday.

BW
@5b4wn

closed 2024-03-29 by ssilverman