ssilverman/QNEthernet #78: QNEthernet 0.28.0 version error message "error: call of overloaded 'EthernetServer(int)' is ambiguous"

nazlyteran, 2024-06-05:
### Discussed in https://github.com/ssilverman/QNEthernet/discussions/75

<div type='discussions-op-text'>

<sup>Originally posted by **ssilverman** May 15, 2024</sup>
## Added
* Added raw frame loopback support and a `QNETHERNET_ENABLE_RAW_FRAME_LOOPBACK` macro to enable.
* Added a fourth step to the _MbedTLSDemo_ example instructions: modify the config.
* New `EthernetFrameClass` functions: `destinationMAC()`, `sourceMAC()`, `etherTypeOrLength()`, and `payload()`.
* Consolidated all the hardware abstraction layer (HAL) functions into one place: `qnethernet_hal.cpp`.
* New `NullPrint` and `PrintDecorator` utility classes in the `qindesign::network::util` namespace.
* Added `driver_is_link_state_detectable()` function to the driver. This is for detecting whether the hardware is able to read the link state.
* Added `EthernetClass::isLinkStateDetectable()` to detect whether the driver is capable of detecting link state.
* Added `setOutgoingDiffServ(ds)` and `outgoingDiffServ()` functions for modifying and accessing the differentiated services (DiffServ) field, respectively, in the outgoing IP header, to `EthernetClient` and `EthernetUDP`.
* Added `EthernetUDP::receivedDiffServ()` for retrieving the DiffServ value of the last received packet.
* Added `EthernetFrameClass::clear()` for clearing the outgoing and incoming buffers.

## Changed
* Updated the Mbed TLS version in the README and comments to 2.28.8 (was 2.28.6).
* Updated `mbedtls_hardware_poll()` in _MbedTLSDemo_ example for other platforms.
* Renamed `QNETHERNET_ENABLE_CUSTOM_WRITE` to `QNETHERNET_CUSTOM_WRITE`.
* Improved the _PixelPusherServer_ example.
* The address-changed callback is now called for independent IP address, netmask, and gateway changes.
* Improved link function documentation for `linkStatus()`, `linkState()`, and `isLinkStateDetectable()`.
* Updated `EthernetClient::setNoDelay(flag)` to return whether successful.
* Add another 2 to `MEMP_NUM_SYS_TIMEOUT` option for mDNS, for a total of an additional 8. Timeout exhaustion was still observed with 6. Why 8 and not 7:
  * https://lists.nongnu.org/archive/html/lwip-users/2024-05/msg00000.html
  * https://savannah.nongnu.org/patch/?9523#comment18
* Updated `EthernetClient::connect()` to return a Boolean value. (The function signatures don't change; they still return an `int`.) This matches the new definition at [Ethernet - client.connect()](https://www.arduino.cc/reference/en/libraries/ethernet/client.connect/).
* Changed `EthernetClient::connectNoWait()` return types to `bool`.

## Fixed
* Improved marking of unused parameters.
* Fixed up use of `__has_include()`, per: [__has_include (The C Preprocessor)](https://gcc.gnu.org/onlinedocs/cpp/_005f_005fhas_005finclude.html)
* In mDNS, ensure there's at least an empty TXT record, otherwise the SRV record doesn't appear.
* Make the alternative `yield()` implementation `extern "C"`.
* Fixed `EthernetUDP` data copy for zero-length outgoing packets. `pbuf_take()` considers NULL data an error, so only copy the data if the packet's size is not zero.

<hr /><em>This discussion was created from the release <a href='https://github.com/ssilverman/QNEthernet/releases/tag/v0.28.0'>0.28.0</a>.</em></div>

ssilverman, 2024-06-05:
Closing because this issue needs more detail. (I’m not sure why the changelog was posted. Is this a quirk of the UI or something?)

ssilverman, 2024-06-05:
Ah.  The text is here: https://github.com/ssilverman/QNEthernet/discussions/75#discussioncomment-9672502

ssilverman, 2024-06-05:
I copied the issue text:

Hi @ssilverman , amazing work with this library. Thank you so much. 
Currently, I'm using it for two projects , one of them is a basic server which after updating the QNEthernet library an instruction  in my program has come up with one error: 
Instruction 
**EthernetServer server(14555);** 
Error:
**error: call of overloaded 'EthernetServer(int)' is ambiguous.** 
I'm wondering why this is happening, and what I need to modify to make it working with the lates library version. 

This is the complete program

```c++
#include <QNEthernet.h>

using namespace qindesign::network;

const unsigned int MAX_MESSAGE_LENGTH = 255;
int size;

IPAddress ip{192, 168, 0, 42};         // device IP
IPAddress sn{255, 255, 0, 0};     // Subnet Mask
IPAddress gw{192, 168, 0, 1};          // Default Gateway

EthernetServer server(14555);    // server on the 14555 port
//===============================================================================
//  Initialization
//===============================================================================

void setup() {
  Serial1.begin(57600);             // gimbal port
  Serial.begin(57600);              // USB Port

  Serial.println(Ethernet.begin(ip, sn, gw) ? "success" : "failure");   // Print if I have connection with the network

  Serial.println(Ethernet.interfaceStatus() ? "UP" : "DOWN");           // preguntar por este codigo que hace ?

  // Show whether a cable is plugged in or not.
  Ethernet.onLinkState([](bool state)
                       { Serial.printf("[Ethernet] Link %s\n", state ? "ON" : "OFF"); });

 // start listening for clients
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());

}

//===============================================================================
//  Main
//===============================================================================

void loop() {

  // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");

     while (client.connected()) {
        serial1Event();

         delay(100);
      static byte message[255]; 
       int size = client.available(); // get the data size
      for (int i = 0; i < size; i++)
      {
        
        message[i] = client.read();
        Serial1.write(message[i]);
      //  Serial.println(message[i], HEX);
      }
      
         
    }

        // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }

}

void serial1Event() {
  
  static byte messageS[MAX_MESSAGE_LENGTH]; //Create a place to hold the incoming message
  while (Serial1.available() > 0)
  {
    
    size = Serial1.readBytesUntil('\n',messageS,255); // Read  message until the character \n
        server.write(messageS, size); // send  message to the client
        
   
  }
  return;
}
```

ssilverman, 2024-06-05:
@nazlyteran could you please post these things:
1. A copy of the error message,
2. Which version of the _QNEthernet_ library you’re using, and
3. The Arduino IDE version you’re using.

ssilverman, 2024-06-05:
This is fixed in the latest code in the repo. To solve for now, do this:

```c++
EthernetServer server(uint16_t{14555});
```

closed 2024-06-05 by ssilverman