ssilverman/QNEthernet #79: Multicast IGMP group timeout

balla94, 2024-07-16:
Hi,

I made a multicast-based audio receiver.  

In the setup() code i call this (but i tried putting the udp.beginMulticast() to the end of the setup() instead of the IP address callback, it changes nothing and tried using the "reuse" version as well) :
`
void setup_ethernet()
{
  uint8_t mac[6];
  Ethernet.macAddress(mac); // This is informative; it retrieves, not sets
  printf("MAC = %02x:%02x:%02x:%02x:%02x:%02x\r\n",
         mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);

  Ethernet.onLinkState([](bool state)
                       {
                         printf("[Ethernet] Link %s\r\n", state ? "ON" : "OFF"); 
                         if(state)
                         {
                            printf("Starting Ethernet with DHCP...\r\n");
  if (!Ethernet.begin()) {
    printf("Failed to start Ethernet\r\n");
  }
  if (!Ethernet.waitForLocalIP(kDHCPTimeout)) {
    printf("Failed to get IP address from DHCP\r\n");
  }
                         } });
  Ethernet.onAddressChanged([]()
                            { 
                              /*configureIP();*/
                              Serial.print("Got IP Address:");
                              Serial.print(Ethernet.localIP());
                              Serial.print("\r\n");
                              udp.beginMulticast(IPAddress(239,0,1,118),8077);
                                 });
  Ethernet.begin();
}

`
Then i call Ethernet.loop() and the following function in each loop cycle:

`void handleUdp()
{
  int packetSize = udp.parsePacket();
  if (packetSize > 0)
  {

    // Read the packet into a buffer
    char packetBuffer[1500];
    int len = udp.read(packetBuffer, 1500);
    if (len > 0)
    {
      // Calculate the number of samples
      int numSamples = len / 4; // Each sample is 4 bytes (2 bytes left + 2 bytes right)

      // Allocate arrays to hold the left and right channel data
      int16_t leftChannelData[numSamples];
      int16_t rightChannelData[numSamples];

      // Extract the left and right channel data
      for (int i = 0; i < numSamples; i++)
      {
        // Extract the left channel 16-bit value
        leftChannelData[i] = (packetBuffer[i * 4 + 1] << 8) | (packetBuffer[i * 4]);

        // Extract the right channel 16-bit value
        rightChannelData[i] = (packetBuffer[i * 4 + 3] << 8) | (packetBuffer[i * 4 + 2]);
      }

      // Pass the left channel data array to queue1.play
      queue1.play(leftChannelData, numSamples);

      // Pass the right channel data array to queue2.play
      queue2.play(rightChannelData, numSamples);
    }
    // Serial.print("Packet contents: ");
    // Serial.println(packetBuffer);
  }
}`

This code works well, it joins to the multicast group, i can see it in the switch as well, but after exactly 4 minutes and 15 seconds (checked with a simple stopwatch), it gets kicked from the IGMP snooping group, it disappears from the switch and stops receiving the stream. 

Do i miss something, or is this a bug? My current workaround is to set static IGMP snooping group in the switch, otherwise it stops every 4:15.

ssilverman, 2024-07-16:
Just to make sure I’m not missing anything, could you format your code so it’s more readable? See here: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

balla94, 2024-07-16:
Sorry, meanwhile i made a pastebin:
[https://pastebin.com/sgkB3tqR](url)

I played with the library a bit, and i found the issue:

For the first time, the igmp timer runs perfectly, but when it reaches zero, it isn't re-triggered anymore. I made a workaround to confirm this, and it works, keeps updating the membership

I modified the original function to this in src/lwip/ipv4/igmp.c:

```
static void
igmp_timeout(struct netif *netif, struct igmp_group *group)
{
  /* If the state is IGMP_GROUP_DELAYING_MEMBER then we send a report for this group
     (unless it is the allsystems group) */
  if ((group->group_state == IGMP_GROUP_DELAYING_MEMBER) &&
      (!(ip4_addr_eq(&(group->group_address), &allsystems)))) {
    LWIP_DEBUGF(IGMP_DEBUG, ("igmp_timeout: report membership for group with address "));
    ip4_addr_debug_print_val(IGMP_DEBUG, group->group_address);
    LWIP_DEBUGF(IGMP_DEBUG, (" on if %p\n", (void *)netif));

    group->group_state = IGMP_GROUP_IDLE_MEMBER;

    IGMP_STATS_INC(igmp.tx_report);
    igmp_send(netif, group, IGMP_V2_MEMB_REPORT);
    igmp_start_timer(group,IGMP_TMR_INTERVAL * 5);
    group->group_state = IGMP_GROUP_DELAYING_MEMBER;
  }
}
```

And added a very ugly debug to this:
```
void
igmp_tmr(void)
{

  struct netif *netif;

  NETIF_FOREACH(netif) {
    struct igmp_group *group = netif_igmp_data(netif);

    while (group != NULL) {
      if (group->timer > 0) {
        group->timer--;
          char out[50];
          sprintf(out,"timer: %d\n",group->timer);
          LWIP_DEBUGF(IGMP_DEBUG, (out));
        if (group->timer == 0) {
          igmp_timeout(netif, group);
        }
      }
      group = group->next;
    }
  }
}
```

ssilverman, 2024-07-16:
Tapping on that link doesn’t take me to the pastebin. May I request that you make this review a little easier? I’m not near my computer (I’m on my phone) and I'm having to do annoying work to just look at the bug.

My specific requests to make it easier for me to help you:
1. Format the code in the first post so it’s displayed as a code block.
2. Make sure any links you add go to where you think they’re going. I shouldn’t have to cut & paste text just to go to a URL.
3. Show the differences for any changes you made (eg. those IGMP functions) so that I can see what you changed, rather than requiring me to open up another tab and then manually compare by switching tabs.

Please make as little work as possible for me to help you. Imagine you’re on a phone when you look at what you’ve written. :)

Does this make sense? I really want to help.

balla94, 2024-07-17:
Sorry, let me try again:

So here is my test code, this runs on a Teensy 4.1 with the SGTL5000 sound card. 
[Pastebin link](https://pastebin.com/sgkB3tqR)

The issue: 
It joins to the multicast group (239.0.1.118) once, then never refreshes the multicast group membership on the network, the switch times out after ~4 minutes and 15 seconds.  Once this time is up, the switch removes the Teensy from the multicast group and the stream stops working on this device.  I don't know why, but if i pull out the ethernet wire and re-attach it, after it gets IP address again, it joins again to the IGMP group, but it does this only once then timeouts again. 

Workaround:
Make static multicast group on the switch or use dumb switch.

What i did:
src/lwip/ipv4/igmp.c: (starting from line 666)
[Direct link to igmp.c](https://github.com/ssilverman/QNEthernet/blob/master/src/lwip/ipv4/igmp.c)

```cpp
static void
igmp_timeout(struct netif *netif, struct igmp_group *group)
{
  /* If the state is IGMP_GROUP_DELAYING_MEMBER then we send a report for this group
     (unless it is the allsystems group) */
  if ((group->group_state == IGMP_GROUP_DELAYING_MEMBER) &&
      (!(ip4_addr_eq(&(group->group_address), &allsystems)))) {
    LWIP_DEBUGF(IGMP_DEBUG, ("igmp_timeout: report membership for group with address "));
    ip4_addr_debug_print_val(IGMP_DEBUG, group->group_address);
    LWIP_DEBUGF(IGMP_DEBUG, (" on if %p\n", (void *)netif));

    group->group_state = IGMP_GROUP_IDLE_MEMBER;

    IGMP_STATS_INC(igmp.tx_report);
    igmp_send(netif, group, IGMP_V2_MEMB_REPORT);
    
    
    /* I've added the following 2 lines and the problem disappeared : */
    igmp_start_timer(group,IGMP_TMR_INTERVAL * 5);
    group->group_state = IGMP_GROUP_DELAYING_MEMBER;
    
    
    
  }
}
```

With the last 2 lines, it starts periodically refreshing the IGMP membership and begins working as expected, but I'm not sure how this should be handled correctly

ssilverman, 2024-07-18:
Thanks for the update. First, a few comments before I forget:
1. It looks like you're calling `Ethernet.begin()` twice.
2. There's no need to call `Ethernet.loop()` inside the main loop() because it's called automatically inside yield(), each time the main loop() completes. I think I see why you're doing it, however, and it looks like it's because there's a potential while loop when there's no link. Here's what I would do: Set a flag in the link listener that's then checked inside the main loop() to do an action.

As I write this, I just noticed that you're also doing a bunch of network setup inside the link listener. (I didn't notice because of the "not optimal" indentation that makes this hard to read, which also makes it look like you're calling `Ethernet.begin()` twice.) Don't do that. Instead, set some variables from inside the listener(s) and check from the main loop().

The IGMP timer is actually called once every 100ms or so, so there's no need to adjust the _igmp.c_ code. I've been using my library for lots of other multicast-based projects and I've never seen this issue. It's very likely because of the program structure and the weird mix of listeners, attempted re-initialization, network calls inside the listener(s), and the while loop inside the main loop().

Could you describe exactly what your app needs to do and maybe I can help restructure?

ssilverman, 2024-07-18:
I tried to build the file as-is and I encountered these problems; it looks like there's some things missing and I don't know which libraries to install or where you've set the values:
1. networking.h
2. TAS2780.h
3. `kDHCPTimeout` isn't defined
4. ArduinoHttpClient.h and ArduinoJson.h aren't needed
5. QNEthernetUDP.h isn't needed because it's already included with QNEthernet.h
6. Defining `LWIP_DEBUG_ENABLED` in the app doesn't do anything

I commented all these out and defined `kDHCPTimeout` to be 15000ms.

ssilverman, 2024-07-18:
I took the liberty of refactoring and modifying the code as I would have started. Note that some names and formatting were changed to match my own preferences. Also, I added an MIT license at the top, mostly to protect myself; I don't have any idea if it works. Think of it as just an illustration.

Gist link: https://gist.github.com/ssilverman/89f5b62e5abdbc8af3a45188537ad302

balla94, 2024-07-18:
Thank you! I will try it in the next hours. 

Meanwhile i made the changes you asked, but unfortunately the same thing happening every time.  

I pushed the [Teensy receiver](https://github.com/balla94/teensy-multicast-receiver) and the [ESP32 transmitter](https://github.com/balla94/esp32-a2dp-to-multicast) code.  

I use D-link DGS-1100-08PV2 switch, but i tried with Mikrotik CSS326 as well with enabled IGMP snooping. 
The transmitter is an Olimex ESP32 Gateway board.

ssilverman, 2024-07-18:
Looks like you don't need `kDHCPTimeout` anymore since your code doesn't use it.

ssilverman, 2024-07-19:
I did some more testing. (Side question: What version of the _QNEthernet_ library are you using?)

My setup:
* eero network
* Teensy 4.1 Ethernet plugged into an eero
* Latest _QNEthernet_ from GitHub (36aa4464c)
* sACNView sending an sACN packet about once every 850ms to universe 1 (multicast 239.255.0.1)
* Code: https://gist.github.com/ssilverman/31295fc40032720dc8e3bf0198bd65a3

Test 1:
1. Start program on the Teensy
2. Start sending packets
3. Observe packet reception for 5 minutes
4. Stop sending packets
5. Wait for 5 minutes
6. Observe no packets for this duration
7. Resume sending packets
8. Observe packet reception for 5 minutes

Test 2:
1. Start program on the Teensy
2. Send no packets for 5 minutes
3. Observe no packets for this duration
4. Start sending packets
5. Observe packet reception for 5 minutes
 
Both tests passed. I suspect it's your router (or something in the lwIP stack that isn't compatible with your router).

balla94, 2024-07-19:
Thanks for your effort!

I use the latest library. 
Meanwhile i did experiments too. 

My observations:
A lot of switches have "unknown multicast flood" enabled by default or running without the IGMP snooping, so i tried things with the following clean setup.

RB1100AHx4 Mikrotik router
- Unknown multicast flood disabled
- IGMP snooping on bridge enabled
- Multicast Querier enabled
- All ports from ether2-ether13 bridged
- The rest is the default quickset config, no firewall or other rules

lwiopts.h
```cpp// Debugging options
#define LWIP_DEBUG
#define LWIP_DBG_MIN_LEVEL LWIP_DBG_LEVEL_ALL
#define LWIP_DBG_TYPES_ON  LWIP_DBG_ON
#define NETIF_DEBUG        LWIP_DBG_ON
#define SOCKETS_DEBUG      LWIP_DBG_ON
#define ICMP_DEBUG         LWIP_DBG_ON
#define IGMP_DEBUG         LWIP_DBG_ON
```

If i connect the ESP32 to ether12 and start the A2DP stream (with the code i've posted above), the port led starts blinking, i see the incoming traffic on the bridge. So the stream is working on 239.0.1.118. 

Test 1: 
Connect Teensy to ether13, music is playing on Teensy, but the router detects that this client is directly attached (L flag in the bridge/ hosts), so this isn't appear on the MDB table.  Music is working, no timeout at all, but i see 
`igmp_timeout: report membership for group with address 239.0.1.118 on if 0x20006e0c`
message just once. 

Test 2: 
ESP32 connected to Mikrotik ether12, stream is active, D-link DGS1100-08PV2 with enabled IGMP snooping connected to Mikrotik ether13. Both leds are blinking (ether13 on mikrotik, and the ether1(uplink) led on the D-link), even if no other devices are connected to the switch, this is okay. 

If i connect a random computer to the switch, it doesn't pass the multicast traffic, so the snooping is okay. If i connect the Teensy to the D-link switch, traffic starts flowing on that port, i get IP address from the mikrotik, the multicast address appears on the Mikrotik MDB table and on the D-link IGMP list as well.  
I get
`igmp_timeout: report membership for group with address 239.0.1.118 on if 0x20006e0c`
in the console once, and the music plays for 4 minutes and 20 seconds. Then the ethernet port on teensy stops blinking. I left it for minutes, but nothing. The other port leds (esp32 and between the router and the switch) keeps blinking.

Test 3:
Same as the above, but i start VLC at the same time with this:
```bash
./vlc udp://@239.0.1.118:8077 --miface=192.168.200.60 --demux=rawaud --rawaud-channels=2 --rawaud-samplerate=44100 --rawaud-fourcc=s16l```

Teensy stops playing after 4:20, but the computer keeps playing. 

Test 4:

I modified óthe igmp.c and ran Test 2 and Test 3 again:

```cpp
static void
igmp_timeout(struct netif *netif, struct igmp_group *group)
{
  /* If the state is IGMP_GROUP_DELAYING_MEMBER then we send a report for this group
     (unless it is the allsystems group) */
  if ((group->group_state == IGMP_GROUP_DELAYING_MEMBER) &&
      (!(ip4_addr_eq(&(group->group_address), &allsystems)))) {
    LWIP_DEBUGF(IGMP_DEBUG, ("igmp_timeout: report membership for group with address "));
    ip4_addr_debug_print_val(IGMP_DEBUG, group->group_address);
    LWIP_DEBUGF(IGMP_DEBUG, (" on if %p\n", (void *)netif));

    group->group_state = IGMP_GROUP_IDLE_MEMBER;

    IGMP_STATS_INC(igmp.tx_report);
    igmp_send(netif, group, IGMP_V2_MEMB_REPORT);
    
    /* Added the following 2 lines: */
    group->group_state = IGMP_GROUP_DELAYING_MEMBER; 
    igmp_start_timer(group,500); 
    
  }
}
```

Result : 
`igmp_timeout: report membership for group with address 239.0.1.118 on if 0x20006e0c`
Messages coming to console every ~20 seconds, and the music keeps playing over 4:20. 

If i enable the timer debug as well, i see the igmp_tmr() always ticks as you mentioned, but the igmp membership timeout gets triggered only once. So the timer is ticking, but the state machine never refreshes the membership if i leave the igmp.c untouched.

ssilverman, 2024-07-19:
I'm not yet convinced that this is the correct fix. I found this:

https://lists.defectivebydesign.org/archive/html/lwip-devel/2016-04/msg00045.html
(Note: The Savannah site where the mailing list is actually hosted seems to be down today.)

This points to the IGMP RFC: https://datatracker.ietf.org/doc/html/rfc2236#section-6
In the state diagram, it says that there is a transition to the "Delaying Member" state only when there's a query from the "Idle Member" state or the "Delaying Member" state. When the timer expires and a group is in the "Delaying Member" state, the state is supposed to transition to "Idle Member".

Basically, I'm not seeing how keeping the state as "Delaying Member" when the timer expires is correct.

balla94, 2024-07-19:
> As far as semantics, this seems like a more correct replacement:
> 
> ```c++
>     igmp_delaying_member(group, IGMP_JOIN_DELAYING_MEMBER_TMR);
> ```
> 
> Note also that the parameter is supposed to be in ticks and not milliseconds.
> 
> Does this line work for you? (i.e. instead of your two added lines)

Yes, it works, but it spams the switch with multiple join requests per second. 

I will read the links in the next hours and doing more tests

balla94, 2024-07-22:
Found it!

The IGMP querier sends out the general membership queries with 0.0.0.0 SRC address, but these packets gets dropped inside the teensy. 
I've enabled 
```cpp #define QNETHERNET_ENABLE_PROMISCUOUS_MODE 1 ```
in src/qnethernet_opts.h and i started to receive these packets and it is now working as expected.

```
igmp_init: initializing
igmp_start: starting IGMP processing on if 0x2000677c
igmp_lookup_group: allocated a new group with address 224.0.0.1 on if 0x2000677c
[Ethernet] Link ON
Starting Ethernet with DHCP...
Failed to get IP address from DHCP
igmp_input: message from 0.0.0.0 to address 224.0.0.1 on if 0x2000677c
igmp_input: General IGMP_MEMB_QUERY on "ALL SYSTEMS" address (224.0.0.1) [igmp_maxresp=96]
igmp_report_groups: sending IGMP reports on if 0x2000677c
igmp_input: message from 0.0.0.0 to address 224.0.0.1 on if 0x2000677c
igmp_input: General IGMP_MEMB_QUERY on "ALL SYSTEMS" address (224.0.0.1) [igmp_maxresp=96]
Initializing LittleFS ...Total space: 131596288 used space: 9437184 used space percentage: 8 %
Audio board begin
igmp_lookup_group: allocated a new group with address 239.0.1.118 on if 0x2000677c
igmp_joingroup_netif: join to new group: 239.0.1.118
Play1
igmp_timeout: report membership for group with address 239.0.1.118 on if 0x2000677c
igmp_input: message from 0.0.0.0 to address 224.0.0.1 on if 0x2000677c
igmp_input: General IGMP_MEMB_QUERY on "ALL SYSTEMS" address (224.0.0.1) [igmp_maxresp=96]
0x2000677c
igmp_input: message from 0.0.0.0 to address 224.0.0.1 on if 0x2000677c
```

ssilverman, 2024-07-23:
Your report was very helpful, thank you. I think I managed to fix the issue. Check out the latest `master`. (Specifically, the fix is in commit b2d34b47f.)

balla94, 2024-07-23:
I just tested it and works perfectly. Thanks for your help!

closed 2024-07-23 by balla94

ssilverman, 2024-07-23:
Thanks for the report and helping diagnose the problem.