ssilverman/QNEthernet #82: v0.29 unusable at 24Mhz and slower than v0.28 at 600Mhz

Mike4U, 2024-08-07:
This is a continuation of a discussion on the teensy forum starting at [here](https://forum.pjrc.com/index.php?threads/new-lwip-based-ethernet-library-for-teensy-4-1.68066/post-347435).

My original post had this physical connection:  teensy->netgearGS305Pswitch->50ft catE cable->verisonMR424WR->wifi->MacAir.  Tried with 2 different teensy setups with several different teensy software and with unix time, same result.

Just tried teensy->verison->Mac.  Same result.

I've tried deleting various cache and even tried with Teenduino 1.8.15 with the same result.  Before I try deleting and reinstalling Arduino 2.3.2 I'm going to wait for maybe a week and see if anyone else has issues.

ssilverman, 2024-08-08:
What happens if you connect the Teensy directly to your computer via Ethernet?

Mike4U, 2024-08-08:
Unfortunately my M2 MacAir only has wireless.  I tried an older computer (2007MacBook) but haven't figured out how to get the direct connection working, both on the MacBook and in the program.  

I did get this response from it with v0.29 :
12
GET / HTTP/1.1
13
User-Agent: curl/7.21.4 (universal-apple-darwin11.0) libcurl/7.21.4 OpenSSL/0.9.8z zlib/1.2.5
14
GET / HTTP/1.1
15
User-Agent: curl/7.21.4 (universal-apple-darwin11.0) libcurl/7.21.4 OpenSSL/0.9.8z zlib/1.2.5

with time curl of:
real 	0m2.342s
user	0m0.012s
sys  	0m0.006s

Then with v0.28:
1
GET / HTTP/1.1
2
GET / HTTP/1.1

with time curl of:
real 	0m0.021s
user	0m0.012s
sys  	0m0.006s

However I couldn't get FireFox to connect to it with either.  Probably my configuration.

verison should have been a Verizon 4 port and wireless router.

ssilverman, 2024-08-08:
I’d like to take that router out of the loop and see what happens. For direct Ethernet, there are those USB-C Ethernet adapters. That’s what I use.

Mike4U, 2024-08-08:
As I said, the last data was a direct connection.  But I got the teensy started on the Verizon, then unplugged the cable and plugged directly into the MacBook.  Amazingly the connection sort of worked except FireFox couldn't find it.

How do I configure my program for no DHCP?

ssilverman, 2024-08-08:
You’re saying this:
> As I said, the last data was a direct connection.

But you said this earlier:
> Unfortunately my M2 MacAir only has wireless. I tried an older computer (2007MacBook) but haven't figured out how to get the direct connection working, both on the MacBook and in the program.

So where did you say you tried a direct connection?

ssilverman, 2024-08-08:
See this part in the AppWithListenersTemplate example: `if (startWithStatic) {`

Mike4U, 2024-08-08:
Sorry about the 'As I said', I'm getting to old and frustrated for this.  Keep getting messed up and confused trying to resolve this.
1. Is there any way for the program to know which version it is compiled with?
2. One of the teensy I'm using is hooked to an LCD display using ILI9341_t3.h.  Thought I'd use it to help debug.  However, as soon as I call tft.println() or any print after something has printed to the serial monitor it freezes.
3. Looking over the AppWithListenersTemplate.  Wife is getting pissed at me.

Mike4U, 2024-08-08:
More exciting info:
Bought a usb-c to ethernet adapter for the M2 MacAir.
Merged the AppWithListenersTemplate into my code which resolved the problem I was having with self assigned ip address.
Results:
1. Mac->USB-C->ethernet cable->teensy    v0.28 ~= v0.29  = 47ms
2. Mac->USB-C->ethernet cable->4port switch->50ft E cable->Verizon->50ft cable->4port->teensy  about the same as 1
3. Mac or phone or iPad wireless ->Verizon->50ft cable->4port->teensy  v0.28 about 51ms, v0.29 in consecutive tries usually something like 83ms, 47ms, 1 second, 2 second .. ~4 seconds.

So something happens with any wireless connection to my router on v0.29

Mike4U, 2024-08-08:
Did an internet speed test:
wireless down 34 up 47 Mbps
wired through 4port etc  down 85  up 119 Mbps

ssilverman, 2024-08-09:
It might be what that router is filtering or something. One of the changes in v0.29.0 was a fix for joining the “all-systems” multicast group in the right spot. (224.0.0.1). It allows the Teensy to respond to some mDNS queries. (Just thinking aloud…)

ssilverman, 2024-08-09:
Side note: I sent you a private message on the forum.

ssilverman, 2024-08-09:
I believe the problem is solved with the latest push. The problematic commit: 291fdddb44e1e25911ab14a2cca58a68abc6ca83

Mike4U, 2024-08-09:
YA !  You the man !  Tried a while on several programs and that fixed it.
Thanks.

closed 2024-08-09 by Mike4U

ssilverman, 2024-08-09:
Thank you for your help.

Mike4U, 2024-08-09:
Looked over the changes in GitHub desktop and can say I have no idea how you figured out that would, work but it does.  It seemed to
have also pep-up some of the UDP interchange between the teensy and a raspberryPi weather station server here.  Didn’t verify that though.

Thanks for the work.

ssilverman, 2024-08-09:
It was basically only processing one packet at a time instead of everything in the buffer. I made that change initially so that any packet data could be examined in one place instead of in each driver.