ssilverman/QNEthernet #93: Bad TCP write performance

spnda, 2025-03-01:
I'm trying to implement a HTTP web server using your (great!) library. One part of this server is allowing file downloads, and I'm using [chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding) since I don't want to use so much memory, and I don't always know the file size beforehand. For this, I currently have this code which works:
```cpp
static constexpr size_t max_buffer_size = TCP_MSS * 4;
const auto buffer = std::make_unique<uint8_t[]>(max_buffer_size);

size_t available = 0;
while ((available = file.available())) {
	static constinit size_t max_chunk_overhead = fmt::formatted_size(
		FMT_COMPILE("{:X}\r\n"), max_buffer_size) + 2;

	available = min(available, max_buffer_size - max_chunk_overhead);

	auto output_it = fmt::format_to(buffer.get(), "{:X}\r\n", available);
	output_it += file.read(output_it, available);
	output_it = std::copy_n("\r\n", 2, output_it);

	client.writeFully(buffer.get(), std::distance(buffer.get(), output_it + 2));
}

static PROGMEM const auto end_chunk = "0\r\n\r\n";
client.writeFully(end_chunk);
```
Essentially, this reads the file in in 5840 byte chunks, and formats it to use the chunked encoding. Finally, for every chunk it does `client.writeFully(...)`.

However, I have observed that this code maxes out at 1.3 to 1.4MB/s. Given that this is a 100Mbit/s connection, getting only 11.2% of the theoretical maximum is quite disappointing. After doing some "debugging" I have found the file reading/formatting to cost around 250us, and the `writeFully` to be around 5000us, and sometimes even more. Again, this is quite disappointing and I don't know how I could improve this. 

Essentially, I am here to ask for any guidance on how to improve performance in such cases to get closer to the theoretical maximum. I ran the iperf server a while back (albeit with a slightly different network setup) and reached around 95-ish Mbit/s. I have also discovered that pining my server takes at between 3 to 8 milliseconds, which seems awful high to me for a local network.

ssilverman, 2025-03-01:
TCP buffers writes until a timeout (about 250ms) or until an internal buffer is full. The solution is to call `flush()` when you want the data to get sent immediately.

See https://github.com/ssilverman/QNEthernet/blob/375c441b852cc861cb82ece0fa90935e23ae0fab/README.md#write-immediacy

spnda, 2025-03-01:
I tried doing `client.flush` after the `client.writeFully`, and also this:
```cpp
elapsedMillis millis_since_start;
[...]
	// lwip only flushes automatically every 250-ish milliseconds.
	if (millis_since_start >= 10) {
		millis_since_start = 0;
		client.flush();
	}
```
Neither of these helped.

ssilverman, 2025-03-01:
I’d just call `flush()` after your write calls, wherever you want data to be sent immediately.

spnda, 2025-03-01:
Okay, but it still doesn't answer the question of how to improve data throughput. With or without I am still getting at most 1.5MB/s and that should be a lot higher. I am usually going to be transferring multiple files in the range of 50-500MB, meaning that would take ages.

[image]

ssilverman, 2025-03-01:
I’d have to see your complete program to see what’s going on. Could you make a minimal program that demonstrates the issue? Also, could it be that the formatting you show above is eating up some time?

spnda, 2025-03-02:
> Is `output_it` dynamically allocated? Could it be that that function does a lot of dynamic allocation and that’s interfering with efficiency? (I’m not familiar with the `fmt` library you’re using, hence my question.)

Nope, none. It always overwrites the data from `buffer` which I allocate once and then keep around. I measured the entire block of the `fmt::format`, `file.read`, and the `memcpy` to be the `250us` I mentioned. `{fmt}` is an absolutely amazing C++ library which actually got put into the stdlib with C++20 and C++23 and it offers Python-like string formatting at much higher speeds than printf and C++ output streams.

> Also, could it be that the formatting you show above is eating up some time?

The formatting, the file reading, and everything else take up 4% of the loops runtime.

> Could you make a minimal program that demonstrates the issue?

I could try essentially copy & pasting parts of my current codebase to create a simple reproduction. It's essentially just a program to upload files from an SD card through a web server, so most of the other relevant code would be "boilerplate".

ssilverman, 2025-03-08:
I found something in ConnectionManager that I’m questioning: “Why did I do that?” I’m going to prepare a branch for you to try out with a change.

ssilverman, 2025-03-08:
Looks like I spoke too soon and got my hopes up. It turns out I didn't find what I thought I found.

spnda, 2025-03-08:
So uh I didn't look into this more, but I am now back at work and having a direct Ethernet connection with my laptop (with a Intel I216 chip) gives me around 7MB/s, compared to the previous 1.4MB/s. At home I connected the teensy to the back of my Fritz!Box 7950, which is a Gigabit port. I don't really know what to make of this information, but if I find the time and perhaps other devices I could see what devices cause performance issues and which don't.

ssilverman, 2025-03-09:
Tuning this stuff can be tricky, and it’s hard for the library writer to have a good set of “generic” options. The good news is, a lot of the TCP stack can be tuned further. I’m glad you saw improved transfer speeds.

ssilverman, 2025-03-11:
For large transfers, I actually recommend against calling `flush()` after every write because the data will get sent when the internal buffer is full and take advantage of larger packets.

closed 2025-03-18 by ssilverman