ssilverman/QNEthernet #94: TCP client close() function question

megax, 2025-03-11:
Hi,

I have a question, hasn't this condition `!conn_->connected` lost its meaning now? It used to work a long time ago, now if the prerequisites don't allow it, it never works. Am I seeing this correctly?
https://github.com/ssilverman/QNEthernet/blob/07b9f0f73d71d6bbd0efa84e0a5a81d6a7685d99/src/qnethernet/QNEthernetClient.cpp#L323
Only abort() would resolve the given pcb in this case.

Sorry if I worded it wrong, my English is poor.

ssilverman, 2025-03-11:
Thank you for the question. Could you clarify what you mean by "used to work" and "never works"? Note that the user could call `close()` at any time, including before or after the connection is actually closed.

megax, 2025-03-14:
I'll try to ask for help with the translation this weekend. I'm experiencing an error that's why I asked a question, because I'm using a mobile router. And in case of a network disconnection, the connection gets stuck. I detect the timeout, I close/abort, and then I keep getting back that the connection is false when connecting. I'm using the connectNoWait option. For some reason, the router is causing problems. For example, if I connect ethernet to the wan port (in this faulty state), all connections are immediately restored. It also occurs on a plain IP basis (omitting host and dns queries). But, as I wrote, I'll ask for help to better describe my problem. I'm gathering evidence as to why the error might occur. (I'm using the latest current master state, but the error also occurs with older versions) Or a solution, how I can try to make the code try to connect again, or to close what may be limiting this.

ssilverman, 2025-03-14:
`connectNoWait()` can return `false` when there's no more connections available. If I had to guess, it sounds like the stack is trying to maintain a connection even when the other side isn't. If you're running out of sockets and nothing's connected, I would just abort() these after some timeout. See: https://github.com/ssilverman/QNEthernet/blob/master/README.md#on-connections-that-hang-around-after-cable-disconnect

Does this section possibly hint at a solution?

megax, 2025-03-14:
I'll try this suggestion. Question, is there a recommendation on how to close the unnecessary Server tcp connection? Or is there a way to delete all attempts in case of such a hang? So, if I don't accept a connection, what is the correct way to close it? It's possible that that's causing some of the problem. But, I'll try abort().

With this, I try to close the "unserved" connections. It works fine in normal cases.
```
                auto randomclient = _server.accept();

                if(randomclient.connected()) {
                    randomclient.close();
                }
```

ssilverman, 2025-03-14:
I'd suggest to use close() as normal, but if, for some reason, the close takes a long time, fall back to doing an abort of these "closed" connections after a timeout, if they're still not closed.

In summery:
1. Close and add to some "pending close" list, with a timestamp (or just keep track of it in some client-tracking object like in my _ServerWithListeners_ example: `ClientState`)
2. Every so often, check that the clients in that list are actually closed
3. If it's been too long and a client still isn't closed, do abort()

megax, 2025-03-18:
Thanks, it seems that the problem was caused by closing the connections. Abort() solved it (only use abort, when timeout disconnect). I have a question, though. Can I somehow check how full the tcp buffer is? If I understand correctly, my connections share a reserved buffer size. If one of them fills up due to an error, the other one will also be disconnected. (It's not on a mobile data connection, but on a local network). Or any tips on whether this could affect it or if I could break something else? I'm just interested in brainstorming. This solution now fixes my problem 95%. Thanks.

ssilverman, 2025-03-18:
Extra information: close() essentially waits for acknowledgment from the other side, using up a connection until the socket is actually closed. abort() just deletes the connection without waiting for any acknowledgement, and frees the connection right away. When all connections are used up, no more can be created or accepted.

Do you mean the receive or the send buffer and what information are you trying to get?

ssilverman, 2025-03-18:
Each TCP socket has its own buffers. You can see how much room is available for writing with the availableForWrite() function.

ssilverman, 2025-03-18:
TCP has all kinds of mechanisms to adjust rates and windows and sizes, etc. It’s designed to adjust to the conditions. For example, when you don’t process received information quickly, the sending side is told about that and will probably slow down. Also, if data isn’t being sent or acknowledged by the remote side quickly, then the local send buffer will get full until it can drain properly.

TCP can be hard to tune; there’s many parameters that can be adjusted. Try experimenting with those, if you like, in lwipopts.h. Eg. the window size, maximum-segment-size, etc. But I would recommend reading more about TCP tuning. (Unfortunately, I don’t have links to share.)

megax, 2025-03-18:
Hm, but it seems that when the buffers are full, the local device disconnects. And it doesn't reconnect until I delete the entire connections. Is this possible? Or am I running into some anomaly? Sorry, I can only describe what I experienced.

First of all, I don't want to change settings. "Tuning" can also have the opposite effect, unfortunately.

In principle, I can solve it "ugly". If I detect connections and interruptions, the local connection is restored immediately after the interruption. Setting the monitoring to about 5s.

However, I am looking at the continuous 0, because if I have many processes it can temporarily fill up. I don't want to get false signals either. That's why I am actually interested in as much information as possible about this. If possible, I will solve it with your solution, which the library knows.

ssilverman, 2025-03-18:
I want to make sure I’m understanding the issues you’re seeing. Are you saying that connections are accepted and that eventually, you can’t accept any more, plus the connections that are already open or somehow frozen?

ssilverman, 2025-03-18:
I’m using the word “frozen” and “stuck” interchangeably. i.e. I’m using the term “connection being frozen” to mean the same thing as “connection being stuck”.

If there are connectivity problems, it’s because TCP is designed to keep retrying things for a while, often using exponential back-off. The stack doesn’t let a connection become available again until it’s truly closed. This means using close() or one of the write functions — which will wait until the other side acknowledges, even if it never does, which means it could be a long time until the connection is freed or written to — or abort() — which just kills the connection and makes it available again for reuse.

In short, I suspect you have low connectivity, so the connection gets “stuck” or “frozen”. close() or writing data will wait a long time in this case, so you need to use abort() if you’ve decided that close() or writing is taking too long.

ssilverman, 2025-03-18:
I’m thinking about a new section that explains all this that I could add to the Readme.

ssilverman, 2025-03-18:
I’m so glad it works better! :)

ssilverman, 2025-03-18:
Just to clarify one thing: where I was only talking about close(), before I started bringing up writing to the connection, I neglected say that writing data has similar problems as close().

I hope that wasn’t too confusing because you were talking about writing and I kept only talking about close(). :)

Thanks for talking through this with me! :)

megax, 2025-03-18:
It wasn't. I understood :) I can answer more difficultly.

closed 2025-03-18 by ssilverman