ssilverman/QNEthernet #96: QNEthernet and rosserial over tcp

tonyspanos, 2025-06-07:
HI

I want to use your library instead of Arduino Ethernet or NativeEthernet, therefore I adapted the ArduinoTCPHardware.h.
My problem is that:
 - I can run a subscriber successfully over TCP
 - I can run a publisher over TCP
 - Once I try to have both, the subscriber runs until the publisher starts. After that, only the publisher works.

Could you please guide me on how to eliminate the possibility of my misimplementation of your library?

Thanks,

<pre>
#ifndef ROS_ARDUINO_TCP_HARDWARE_H_
#define ROS_ARDUINO_TCP_HARDWARE_H_

#include <Arduino.h>
#if defined(ESP8266)
  #include <ESP8266WiFi.h>
#elif defined(ESP32)
  #include <WiFi.h> // Using Espressif's WiFi.h
#else
  #include <SPI.h>
  #include <QNEthernet.h>
#endif

using namespace qindesign::network;

class ArduinoHardware {
public:
  ArduinoHardware()
  {
  }

  void setConnection(IPAddress &server, int port = 11411)
  {
    server_ = server;
    serverPort_ = port;
  }

  IPAddress getLocalIP()
  {
#if defined(ESP8266) or defined(ESP32)
    return tcp_.localIP();
#else
    return Ethernet.localIP();
#endif
  }

  void init()
  {
    if(tcp_.connected())
    {
      tcp_.stop();
    }
    tcp_.connect(server_, serverPort_);
  }

  int read(){
    if (tcp_.connected())
    {
      if(tcp_.available())
      {
          return tcp_.read();
      }
    }
    else
    {
      tcp_.stop();
      tcp_.connect(server_, serverPort_);
    }
    return -1;
  }

  void write(const uint8_t* data, int length)
  {
    tcp_.writeFully(data, length);
  }

  unsigned long time()
  {
    return millis();
  }

  bool connected()
  {
    return tcp_.connected();
  }

protected:
#if defined(ESP8266) or defined(ESP32)
  WiFiClient tcp_;
#else
  EthernetClient tcp_;
#endif
  IPAddress server_;
  uint16_t serverPort_ = 11411;
};

#endif

</pre>

ssilverman, 2025-06-07:
On which hardware will you be running this?

tonyspanos, 2025-06-11:
Just found it.

The default 
  //typedef NodeHandle_<ArduinoHardware> NodeHandle; // default 25, 25, 512, 512

has a pair of 512 Input/Output buffers, that were apparently small. I replaced it with 2048 bytes each and it works like a charm.

Thanks for the awesome library!

closed 2025-06-11 by tonyspanos